pypilot/pypilot #153: Idea: Truncate BBN OS to run just pypilot on pi zero 2

mgrouch, 2022-09-20:
Idea: Truncate BBN OS to run just pypilot on pi zero 2.

I can create a separate image for BBN OS by truncating it to run just pypilot, and pi zero 2 hardware.
So it's going to be headless. But based on Bullseye and 64 bit. (can be 32 bit with more work)

I've played with it and made experimental images for pi zero 2 before based on buster.

Should I attempt? I do not have pypilot hardware to test. But given that some people use pypilot on BBN regular version
it shouldn't be difficult to cut it down. I'd need to some help testing it though.

Thanks

seandepagnier, 2022-09-20:
what is the purpose?   There is already tinypilot for the pi zero 2 using tinycore 13.

If you can get BBN to work on other sbc, and especially if you can boot into a ramdisk environment (can remove sd card after boot without issues) that would be really interesting.

closed 2022-09-21 by mgrouch

mgrouch, 2022-10-01:
Which SBC? I think I could do Armbian for some other SBC
For pypilot only or for whole thing?

Thanks