pypilot/pypilot #155: Zeroconf' object has no attribute '_handlers_lock'

mgrouch, 2022-09-23:
```

Sep 09 18:23:27 lysmarine pypilot[1543]: Exception in thread zeroconf-Engine:
Sep 09 18:23:27 lysmarine pypilot[1543]: Traceback (most recent call last):
Sep 09 18:23:27 lysmarine pypilot[1543]:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
Sep 09 18:23:27 lysmarine pypilot[1543]:     self.run()
Sep 09 18:23:27 lysmarine pypilot[1543]:   File "/usr/lib/python3/dist-packages/zeroconf/__init__.py", line 1228, in run
Sep 09 18:23:27 lysmarine pypilot[1543]:     reader.handle_read(socket_)
Sep 09 18:23:27 lysmarine pypilot[1543]:   File "/usr/lib/python3/dist-packages/zeroconf/__init__.py", line 1301, in handle_read
Sep 09 18:23:27 lysmarine pypilot[1543]:     self.zc.handle_response(msg)
Sep 09 18:23:27 lysmarine pypilot[1543]:   File "/usr/lib/python3/dist-packages/zeroconf/__init__.py", line 2501, in handle_response
Sep 09 18:23:27 lysmarine pypilot[1543]:     with self._handlers_lock:
Sep 09 18:23:27 lysmarine pypilot[1543]: AttributeError: 'Zeroconf' object has no attribute '_handlers_lock'

```

seandepagnier, 2022-09-24:
what version of zeroconf have you installed?

mgrouch, 2022-09-24:
Looks like 0.26.1 gets installed

seandepagnier, 2022-09-24:
I have 0.24...

what if you just run python3:

>>> import zeroconf

Do you get the error as well?

mgrouch, 2022-09-24:
> what if you just run python3:
>  `>> import zeroconf`

no error if I do that.

seandepagnier, 2022-09-24:
This exception is thrown in a different thread without any reference to my own code.  I am pretty sure it is a zeroconf issue.

Can you try version 0.24 of zeroconf?   When did this error first appear? What did you change?  obviously zeroconf was working in the previous version correct?