pypilot/pypilot #156: Failed to open settings file for saveDetected/saveUsing

mgrouch, 2022-09-23:
```
Sep 14 09:17:15 lysmarine pypilot[1408]: Settings file not found. Using defaults and creating settings file
Sep 14 09:17:15 lysmarine pypilot[1408]: Failed to open settings file for saveDetected MPU9250/MPU9255 at standard address
Sep 14 09:17:15 lysmarine pypilot[1408]: Failed to open settings file for saveUsing fusion algorithm Kalman STATE4
Sep 14 09:17:15 lysmarine pypilot[1408]: min/max compass calibration not in use

```

seandepagnier, 2022-09-24:
probably because it doesnt have permission to write to whatever the current directory is

mgrouch, 2022-09-24:
yep

mgrouch, 2022-09-24:
Actually the issue is that pypilot creates file in /home/pypilot/.pypilot

with 644 permissions which do not give 'pypilot' group write access.

So if pypilot is running as user pypilot and user running GUI tools is in group 'pypilot'
it doesn't give 'write' access to GUI utilities.

mgrouch, 2022-09-24:
I had to set special

setfacl -d -m g:pypilot:rw /home/pypilot/.pypilot

mgrouch, 2022-09-24:
not pnpm issue in my case.
truned out to be broken package 'underscore'

mgrouch, 2022-09-24:
Also it's looking for RTIMULib.ini

seandepagnier, 2022-09-24:
This shouldn't matter since /home/pypilot is different from /home/pi right?

So the user should have access to /home/pi/.pypilot for gui configuration files which dont do much anyway besides set the preferred host name.    They shouldnt need to access /home/pypilot/.pypilot correct?

I dont think you should have to change the group permissions.   Maybe the working directory when pypilot service is run needs to be /home/pypilot where it can write temporary files etc??   I'm not sure what is going on:  it will create RTIMULib.ini if it doesn't exist.

mgrouch, 2022-09-24:
pypilot creates that RTIMULib.ini in /
in my case.
So that didn’t work for me when it is running as ‘pypilot’ user

I’ve fixed for BBN OS in default branch.

seandepagnier, 2022-09-24:
Because pypilot service was launched as user pypilot with working directory of / right?  

Your "fix" is not the right one.   You should set the working directory to /home/pypilot in the systemd script file in my opinion, and not change permissions, and certainly not create files in /

mgrouch, 2022-09-24:
Actually for daemon processes / being working directory is kind of convention.
Reason for that is that it allows un-mounting partitions without stopping daemons.

Is any code in pypilot assumes working directory to be home?

Thanks

mgrouch, 2022-10-05:
It was issue with permissions and working directories in BBN OS.

closed 2022-10-05 by mgrouch