pypilot/pypilot #160: pypilot_calibration utility doesn't load previously stored values after linux reboot

mgrouch, 2022-09-26:
heading offset gets reset
wind sensor offset is loaded as 0
locked accelerometer calibration points are not loaded into pypilot_calibration (shows as default)

There are also some errors like this:
`
Unimplemented OBJ format statement 's' on line 's off'
`

Thanks

seandepagnier, 2022-09-26:
that is a warning, not an error.

What are the values stored in pypilot.conf?

seandepagnier, 2022-09-26:
It is odd there is no imu.compass.calibration

The settings are not stored to disk immediately.   Perhaps this would be nice for values that don't get updated like amp hours or runtime, but it is to avoid constantly writing the file.    If you change the offset and wait until the change is written to pypilot.conf it should be there on reboot.

The other thing is.. stopping pypilot with systemctl stop should flush changes to pypilot.conf, and this should run at reboot, so I am not quite sure what is going wrong here unless you power cycled.

Anyway, no the wind sensor height is not even proven/tested very well so it is 'hidden' for now.

mgrouch, 2022-09-26:
ALso locking compass calibration seems has no effect. It still comes back as unlocked.

mgrouch, 2022-09-26:
Hmm. How a regular user would know to wait?
Usual practice after altering the configuration of a service is to bounce it and see config taking effect.

mgrouch, 2022-10-05:
I think this can be closed. as long as docs say you need to wait for pypilot to stored config.

seandepagnier, 2022-10-05:
it should store the config if the service is stopped.

closed 2022-10-24 by mgrouch