pypilot/pypilot #162: failed to decode data socket! imu pipe[0] Expected object or value

mgrouch, 2022-09-27:
```
Sep 27 14:22:44 lysmarine pypilot[615]: failed to decode data socket! imu pipe[0] Expected object or value
Sep 27 14:22:44 lysmarine pypilot[615]: line {"timestamp":351.558082997,"fusionPoseValid":true,"fusionPose":[0.16661711037158966,0.010481907986104488,-0.021904248744249344],"fusionQPoseValid":true,"fusionQPose":[0.9964536428451538,0.08326328545808792,0.004311120603233576,-0.011349855922162533],"gyroValid":true,"gyro":[-0.00014778412878513336,-0.000482240691781044,-0.0003225398249924183],"accelValid":true,"accel":[-0.013293457217514515,0.1676025390625,0.984728991985321],"compassValid":true,"compass":[34.69593811035156,7.265625476837158,38.671878814697266],"pressureValid":false,"pressure":NaN,"temperatureValid":false,"temperature":NaN,"humidityValid":false,"humidity":NaN,"accel.residuals":[0.0028117410838603973,-0.0017644017934799194,0.0013682842254638672]}

```

I think that error comes up during calibration

Thanks

seandepagnier, 2022-09-28:
This is interesting...   What hardware is connected?   Which IMU do you have?

mgrouch, 2022-09-28:
Raspberry pi4, Bullseye, 64-bit,
IMU:

Waveshare 10 dof C.
https://www.waveshare.com/10-dof-imu-sensor-c.htm

It has 

    MPU9250: 3-axis gyroscope, 3-axis accelerometer, and 3-axis compass/magnetometer
        Internal Digital Motion Processing™ (DMP™) engine, offloads complicate fusion calculation, sensor synchronization, gesture recognition, etc.
        Comparing with MPU6050, it is lower power consumption, and more suitable for wearable devices
    BMP280: barometric pressure sensor
        Embedded temperature sensor for temperature compensation
        Comparing with BMP180, it is lower power consumption, and higher performance
    You can get the 10 DOF data through I2C interface

connected via i2c

USB GPS stick feeding GPSD.

```
Sep 28 13:10:22 lysmarine pypilot[57145]: autopilot start 51951.396775715
Sep 28 13:10:22 lysmarine sudo[57157]:  pypilot : PWD=/home/pypilot ; USER=root ; COMMAND=/usr/bin/chrt -pf 2 57155
Sep 28 13:10:22 lysmarine sudo[57154]:  pypilot : PWD=/home/pypilot ; USER=root ; COMMAND=/usr/bin/chrt -po 0 57152
Sep 28 13:10:22 lysmarine pypilot[57155]: Settings file RTIMULib.ini loaded
Sep 28 13:10:22 lysmarine pypilot[57155]: Detected MPU9250/MPU9255 at standard address
Sep 28 13:10:22 lysmarine pypilot[57155]: Using fusion algorithm Kalman STATE4
Sep 28 13:10:22 lysmarine pypilot[57155]: min/max compass calibration not in use
Sep 28 13:10:22 lysmarine pypilot[57155]: Using ellipsoid compass calibration
Sep 28 13:10:22 lysmarine pypilot[57155]: Using accel calibration
Sep 28 13:10:22 lysmarine pypilot[57145]: loading servo calibration /home/pypilot/.pypilot/servocalibration
Sep 28 13:10:22 lysmarine pypilot[57145]: WARNING: using default servo calibration!!
Sep 28 13:10:22 lysmarine pypilot[57145]: Available Pilots: ['wind', 'absolute', 'basic']
Sep 28 13:10:22 lysmarine pypilot[57145]: warning: failed to open special file /dev/watchdog0 for writing
Sep 28 13:10:22 lysmarine pypilot[57145]:          cannot stroke the watchdog
Sep 28 13:10:22 lysmarine sudo[57163]:  pypilot : PWD=/home/pypilot ; USER=root ; COMMAND=/usr/bin/chrt -pi 0 57152
Sep 28 13:10:22 lysmarine sudo[57169]:  pypilot : PWD=/home/pypilot ; USER=root ; COMMAND=/usr/bin/chrt -pf 1 57145
Sep 28 13:10:22 lysmarine pypilot[57155]: MPU-925x init complete
Sep 28 13:10:23 lysmarine pypilot_web[607]: found pypilot 127.0.0.1 23322
Sep 28 13:10:25 lysmarine sudo[57183]:  pypilot : PWD=/home/pypilot ; USER=root ; COMMAND=/usr/bin/stty -F /dev/ttyLYS_gps_0
Sep 28 13:12:09 lysmarine pypilot[57161]: signalk process 57161

```

seandepagnier, 2022-09-28:
what is happening is...  the actual value is uninitialized and somehow on your system it is (by chance?) a NaN rather than some other random value and the ujson cannot handle it as NaN is not in the json standard.

I have simplified my version of RTIMULib2 to omit the pressure data since it is not used by pypilot anyway and this driver doesnt read it.   Its still available with a different method.   This should solve the problem.   Thanks for reporting it.

closed 2022-10-04 by mgrouch