pypilot/pypilot #164: Question about storing heading offset. Posiible way to improve

mgrouch, 2022-10-02:
In my understanding if user enters heading  offset for IMU as zero, pypilot will try to calculate it based on GPS COG, POS
magnetic variation for position. SOG should be reasonably high. Currents and leeway are not taken into account.

What happens after power off? Is this calculated heading offset stored on file system and used immediately after next power on?

It looks to me like it doesn't.

So if I power on and my heading offset is set to zero I would need to gain some speed on a boat before it will
calculate a new reasonably accurate number.

Is my understanding correct?

Can this previously calculated value for heading offset be stored and used on next power on?
So the heading will be at least more reasonable before boat gains speed?

Thanks
--MG

seandepagnier, 2022-10-02:
It does not yet try to calculate the heading offset, and even if it did, it would only do it via a user interface button similarly to "level" in calibration to ensure there is no leeway and no currents.

RTIMULib.ini is not really used, but it can help the library boot faster since it doesnt need to perform imu detection

closed 2022-10-02 by seandepagnier

mgrouch, 2022-10-02:
Thanks for clarifying it!