pypilot/pypilot #165: pypilot/arduino/motor/Makefile doesn't work with Arduino IDE 1.8.19

mgrouch, 2022-10-04:
I've made new Makefile

https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/pypilot/arduino/motor/Makefile

Also the script to build avrdude:

https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/avrdude/avrdude-install.sh

thanks

seandepagnier, 2022-10-05:
Thanks for this.   I learned about avrdude the hard way, and informed the developers, so I hope the special script is not needed forever.. not sure if you saw this:
https://github.com/avrdudes/avrdude/issues/1107

I have updated the Makefile in pypilot/arduino/hat

I received notice that the serial port does not work with the new arduino which the motor controller uses, so I am hesitating to merge this Makefile.   Once I verify the new arduino can build working motor controllers I will update it.

mgrouch, 2022-10-05:
I've made more fixes in https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/pypilot/arduino/motor/Makefile

Also BBN OS from 2022-10-05 version (and later) includes corrected avrdude binary.

2022-10-05 has Arduino 1.8.19 too and should be able to compile and load MCU firmware for pypilot
without installing anything additionally (except corrected Makefile).

seandepagnier, 2022-10-05:
It probably is fine, but there is extensive testing compiling with the previous compiler and arduino library.

I will continue to use this to produce the binary for my products, as the motor controller is essentially finished anyway and known to be reliable.

I will include this makefile for now as an optional addition until it is proven.   Users can use make -f

closed 2022-10-05 by seandepagnier