pypilot/pypilot #166: command line utility which would check the hardware (make/model/etc) and load appropriate .hex firmware into MCU from some git repository would be nice

mgrouch, 2022-10-05:
command line utility which would check the hardware (make/model/etc) and load appropriate .hex firmware into MCU from some git repository would be nice

thanks

seandepagnier, 2022-10-05:
For the motor controller,   there is no way to detect the hardware as there is just a serial connection, and if the hardware is homemade there is no telling what it is.

All of my motor controllers can use exactly the same firmware, as it supports different options through a few io pins that are tied to ground or left floating (pullup) to determine the capabilities.

As for the pypilot hat..   There is already some code in pypilot to automatically load the hex, but it is disabled at the moment

mgrouch, 2022-10-05:
can be closed then

closed 2022-10-05 by mgrouch

mgrouch, 2022-10-05:
Might few words in the documentation on 

1. Which MCU are supported by motor controller
2. Where to download .hex file
3. What is the command to load .hex file into MCU

Thanks

seandepagnier, 2022-10-05:
The commands to load are in the Makefile.

I do not distributed the compiled hex.

The normal mcu is atmega328