pypilot/pypilot #168: Can IMU detect waves height and report it?

mgrouch, 2022-10-06:
Can IMU detect waves height and report it?

Thanks

mgrouch, 2022-10-06:
I wonder if high precision barometric sensor like DPS310  can measure wave heights.

seandepagnier, 2022-10-06:
It cannot without machine learning and/or in conjunction with other sensors.    The accelerometer is a better bet for this.

mgrouch, 2022-10-06:
If the wave length is longer than ship I think is pretty easy from accelerometer.
Double integral of vertical acceleration gives height.
Period of wave is more tricky. Because you face waves with an angle. 
So there is period when pitch goes from 0 to another zero.
And there is period when heel goes from max to min.
Would be wave period then square root of sum of squares of those two periods?

Thanks

mgrouch, 2022-10-24:
Looks like dps310 pressure sensor is quite
promising for it. 

Here is demo:

https://twitter.com/kogakudanshi/status/1538601537483972610?lang=en

seandepagnier, 2022-10-25:
because of dynamic pressure changes it would be difficult.   Simply having the sensor inside the boat will greatly affect this, as you are trying to detect very slight pressure differences.   The sensor is capable of this resolution, but at sea, the air pressure is turbulent both inside and outside the boat due to sails and boat structure and so forth.   So a different course would yield different results.   Really nothing like air in a still room.

  Similarly it would be possible to train a model to determine wave height from a video feed looking at waves, but either way it is a complex problem.

  What could work is to build some sort of model that allows computing wave height from a pressure sensor, but I still think accelerometers are more viable.

mgrouch, 2022-10-25:
If pressure sensor is in a box with a small hole inside the boat amidship away from airflow?
I doubt it an issue. 

Noise of second integral of accelerometer will be worse.

mgrouch, 2022-10-25:
And another env sensor Lightning Detector - AS3935
I'm sure you will be reluctant to add it on your hat.
I have standalone version https://www.amazon.com/AcuRite-02020-Portable-Lightning-Detector/dp/B00EO1H3X8
and it works fine.
But if it's integrated into boat networks it would be more useful.

mgrouch, 2022-10-26:
For wave height with DPS310 sensor

this code works very nice:

https://github.com/bareboat-necessities/RaspberryPi_DPS/blob/master/Examples/TempPressValues_310.py

rgleason, 2022-10-26:
I don't believe it will predict the higher "rogue" (rollover) waves that are so dangerous to smaller boats.

mgrouch, 2022-10-26:
> I don't believe it will predict the higher "rogue" (rollover) waves that are so dangerous to smaller boats.

That's not the point. At least it gives some sea state measurement metric.

mgrouch, 2022-10-27:
yes, calculating it from IMU would be interesting next step

mgrouch, 2022-10-27:
juangallostra/AltitudeEstimation code seems more suitable for drones

In case of sea waves (as you also pointed out) we know few more additional preconditions which should allow
building a better algorithm

- Average vertical velocity is zero
- Average vertical position is zero
- Sea level pressure/temperature is measured (as opposed to a constant in juangallostra/AltitudeEstimation code)

I like that it has ready to use method for vertical acceleration
`altitude.getVerticalAcceleration() // get estimated vertical acceleration in m/s^2
`

can something like that be added into pypilot? 

imu.verticalAcceleration

currently it has only raw readings from accelerometer XYZ. There is no fused value for vertical acceleration.

Thanks

mgrouch, 2022-10-27:
Actually after reading more in the subject,
wave surface is actually the surface of equal pressure. So basically barometer is useless for measuring wave height. It will pick up only noise from the sensor. It’s then even useless for fusion with IMU data.

mgrouch, 2022-10-31:
I've started working on some code for wave height estimation from IMU. (Unfinished)
I'm basing it on https://github.com/juangallostra/AltitudeEstimation/tree/master/extras

I threw out complimentary pressure sensor fusion as it is useless due to waves changing pressure around them.
The plan is:

Calculate sea wave height from IMU

Algorithm description

-  Project acceleration to Earth center to get vertical acceleration
-  Use Kalman filter to adjust vertical acceleration
-  Calculate integral for vertical velocity and second integral for height
-  Adjust interpolation to zero height when acceleration is low
-  Capture min and max height over periods of time and report difference as wave height

TODO:

    generate NMEA 0183 sentence for wave height
    run this code as part of pypilot server

MWH - Wave Height NMEA 0183 Approved by the NMEA 0183 Standard Committee as of October 1, 2008

`$--MWH,x.x,f,x.x,M*hh Wave height, meters Wave height, feet`

https://github.com/bareboat-necessities/wave_height_from_IMU

I need to replace ZUPT logic with something else. 

I'll try using same approach for adjusting accumulated integrals error as here https://github.com/ryanbarna/waveHeight
(If acceleration lower than a threshold then set height to zero)

The code is in python. 

I'll need to throw out ZUPT, replace reading input from serial to some data stream (pypilot itself, or RTIMULib2?)
Make sure projection for vertical acceleration vector works properly, etc.

rgleason, 2022-11-01:
Significant wave height SWH is mean wave height (trough to crest) of the highest third of the waves.
We can be assured there will be waves about 2 x SWH, and likely even higher. wave frequency of 2xSWH waves will be 1:1000 -1:3000.  Assume a 15-second wave period and a 2xSWH will be seen every 4 to 12-1/2 hours.

Beaufort Gale Force 8 (34 to 40 knots) waves will be 18-25 feet so 2xSWH=40-50' waves.
"Adlard Coles’ Heavy Weather Sailing"   Model testing.. "when the breaking wave was 30 percent of the hull length high, from trough to crest, it could capsize some yachts, while waves to a height of 60 percent of the hull length comfortably overwhelm all of the boats we tested."

Force 8 gale has waves capable of rolling all yachts under 80-feet long.

If you get SWH  working right, you should have a rollover warning, using boat length, so that a drogue can be deployed. Perhaps it should be a part of watchdog_pi.

mgrouch, 2022-11-01:
Found better method. 
not using pressure. 
accounts for zero average displacement by including third integral into Kalman filter. 
specifically designed for waves. 

https://www.researchgate.net/profile/Mehdi-Hendijanizadeh/publication/264713649_A_Novel_Kalman_Filter_Based_Technique_for_Calculating_the_Time_History_of_Vertical_Displacement_of_a_Boat_from_Measured_Acceleration/links/53ec88db0cf24f241f1584c5/A-Novel-Kalman-Filter-Based-Technique-for-Calculating-the-Time-History-of-Vertical-Displacement-of-a-Boat-from-Measured-Acceleration.pdf?origin=publication_detail

seandepagnier, 2022-11-01:
> Actually after reading more in the subject, wave surface is actually the surface of equal pressure. So basically barometer is useless for measuring wave height. It will pick up only noise from the sensor. It’s then even useless for fusion with IMU data.

I was not able to explain my thought but agree.   In fact if you did analyze a barometer reading you would find only noise and it could potentially be possible to predict wave height from the nature of the noise,  but actual pressure changes would not work since as you say, the wave surface has equal pressure, and the pressure sensor is a poor choice.    It would be worse than training a webcam to detect wave height.

mgrouch, 2022-11-02:
> I was not able to explain my thought but agree. In fact if you did analyze a barometer reading you would find only noise and it could potentially be possible to predict wave height from the nature [...]

I'm trying to understand how exactly they use third integral being zero in Kalman filter in this article:

[https://www.researchgate.net/profile/Mehdi-Hendijanizadeh/publication/264713649_A_Novel_Kalman_Filter_Based_Technique_for_Calculating_the_Time_History_of_Vertical_Displacement_of_a_Boat_from_Measured_Acceleration/links/53ec88db0cf24f241f1584c5/A-Novel-Kalman-Filter-Based-Technique-for-Calculating-the-Time-History-of-Vertical-Displacement-of-a-Boat-from-Measured-Acceleration.pdf](https://www.researchgate.net/profile/Mehdi-Hendijanizadeh/publication/264713649_A_Novel_Kalman_Filter_Based_Technique_for_Calculating_the_Time_History_of_Vertical_Displacement_of_a_Boat_from_Measured_Acceleration/links/53ec88db0cf24f241f1584c5/A-Novel-Kalman-Filter-Based-Technique-for-Calculating-the-Time-History-of-Vertical-Displacement-of-a-Boat-from-Measured-Acceleration.pdf)

If you understand can you help me with it?

Thanks

mgrouch, 2022-11-04:
I’ve implemented Kalyan filter to integrate acceleration into wave height with drift correction for integration with noise in input. 

Here is the code. It also draws plots. 

https://github.com/bareboat-necessities/wave_height_from_IMU/blob/main/wave_height_from_IMU.py

seandepagnier, 2022-11-05:
I will study it.   I have also previously implemented a kalman filter to blend gps with inertial sensors.   It provides high rate position and speed estimation and gives a more accurate course/speed.    I implemented to support '3D' but its not tested or proven and needs to force altitude to zero among other things.    It could be interesting to run your implementation in parallel and compare.    It should estimate more than wave height.     It is also impossible to estimate wave height perfectly just from this as you know a very large ship will not move up and down much from small waves.

https://github.com/pypilot/pypilot/blob/master/pypilot/gps_filter.py

  I think the same kalman filter can do both and would be less processing power.      Once it is all working and proven it can also be used by pilot algorithms ( sea state adjustments,  optimizing VMG and so forth)    I will also profile to determine if the filter should be implemented in c++ to reduce cpu usage  since it can really use <1% cpu for this and the pi zero is currently using almost 10%.   I want to maintain at least 50% cpu idle in normal circumstances.

I am really impressed you implemented this,  please give me some time to study, test and reply.

rgleason, 2022-11-05:
What is Kalyan?

mgrouch, 2022-11-05:
> What is Kalyan?

It's a typo. Kalman

mgrouch, 2022-11-05:
> I will study it. I have also previously implemented a kalman filter to blend gps with inertial sensors. It provides high rate position and speed estimation and gives a more accurate course/speed. I [...]

There is wave.m - MatLab file to generate test and reference data.
I've tested on few files. One is acceleration with no noise and no offset where velocity and acceleration 
generated by direct differentiation (first and second) of displacement function. Sampling at 100Hz
Another one same with added noise to acceleration and added small offset.

There are two python scripts one is direct summation integral Kalman and another one with altitude adjustment
based on zero third integral assumption (as in the article).

Having reference data for position and velocity it's easy to play with data and different input variances to see how
algorithms behave. The scripts draw charts.

I've now also tested with IMU data collected from my iphone running up and down stairs and also waving it
up and down. Sampling acceleration Z (trying not to rotate it) at 100Hz and at 50Hz (1 min samples). A bit dirty experiment
because I just used Z accel instead of proper vertical acceleration adjusted to phone rotations.
Even with that it gave quite good results.

mgrouch, 2022-11-05:
Now that it's possible to evaluate vertical velocity it would be interesting to compare how pitch follows it.
Pitch is measured by gyro and magnetometer I'd guess and vertical velocity is from accelerometer.
They should be good metrics to compare as they (I think) should follow each other closely.
Going up - bow up, going down - bow down.

mgrouch, 2022-11-05:
Also few observations on algorithm behavior.

If mean(Acceleration) != 0 (you have constant noise offset) It's going to underestimate wave height. It's going to flatten it on one
of the sides. 

If PosIntegral_Trans_Variance is set too low then it's going to underestimate wave height.

If PosIntegral_Trans_Variance is set too high then it's going to overestimate few first wave heights before converging.
It's going to be more sensitive to noise in acceleration signal

seandepagnier, 2022-11-06:
A few things:

1)   The phase difference between pitch and vertical acceleration will depend on course.   Then a different sea state can completely change it.   It depends on the boat too.    Consider a 5 meter swell against direction of travel and 2 meter swell in the direction.    The boat sinks into the 2 meter swell more, but surfs it.   The larger 5 meter swell has a very long period and doesn't slow the boat much but the vertical acceleration is fully expressed from it.

2)   I still am not sure you can accurately measure wave height.    A small bouy may be closer and more sensitive or it can be compensated, but a boat, as it gets larger becomes unaffected by smaller waves,  and although it is possible to compensate for this, a specific compensation is needed for every boat in every orientation.    At best this will give a rough estimation of wave height.

mgrouch, 2022-11-07:
I propose adding these NMEA sentences. They can be allowed optionally as Kalman filter option too.

```

    
    $APXDR,A,-0.087,M,VDIS*<checksum>
    $APXDR,A,0.08,,VVEL*<checksum>
    $APXDR,A,-0.087,,VACC*<checksum>
    $APMWH,3.44,f,1.05,M*<checksum>
    
    VDIS - vertical displacement (m)
    VVEL - vertical velocity (m/s)
    VACC - vertical acceleration (m/s^2)
    $--MWH - wave height (feet and meters)

```

Wave height can be reported with 1-2 min frequency and others with same frequency as roll and pitch.

rgleason, 2022-11-08:
360 Lidar, https://velodynelidar.com/products/alpha-prime/
would it work with waves?

mgrouch, 2022-11-08:
@seandepagnier 

> I think the same kalman filter can do both and would be less processing power. Once it is all working and proven it can also be used by pilot algorithms ( sea state adjustments, optimizing VMG and [...]

Have you considered using rust instead of C++?

There are nice libraries:

https://crates.io/crates/rppal
https://crates.io/crates/datafusion_imu
https://github.com/rbagd/rust-linearkalman
https://github.com/rust-embedded/embedded-hal

and many more

Thanks

seandepagnier, 2022-11-08:
I am not opposed to rust, but I have no experience and used c++ a lot.

For small modules that implement things like kalman filters there is not a significant advantage to using rust over c++...

mgrouch, 2022-11-09:
@seandepagnier 

> I am not opposed to rust, but I have no experience and used c++ a lot.
> 
> For small modules that implement things like kalman filters there is not a significant advantage to using rust over c++...

Well. I've took pypilot code and transpiled it into rust using

https://github.com/konchunas/pyrs

The script used to transpile it is here: https://github.com/bareboat-necessities/pypilot-pyrs/blob/main/how-it-was-made.sh

and the log: https://github.com/bareboat-necessities/pypilot-pyrs/blob/main/how-it-was-made.log

In the next commit I've applied code formatting using IntelliJ rust plugin.

You can see the result here: https://github.com/bareboat-necessities/pypilot-pyrs

I think it's a good start for pypilot in rust.

Thanks

seandepagnier, 2022-11-10:
This is something different entirely but interesting!    pypilot is written python to make it easier to maintain,  but  if it can be compiled to another lower level language like rust, it could improve performance or reduce cpu usage during runtime.  The compiler you used even admits to not being optimal and potentially crashing, and failed to compile several files.

Judging by the produced output which is almost twice as many lines and more difficult to maintain, it is better to work in python for several reasons.

   A different possibility that I thought you were considering was calling smaller modules written in rust (rather than c++) from python for performance reasons like implementing a kalman filter.

regardless

mgrouch, 2022-11-13:
> This is something different entirely but interesting! pypilot is written python to make it easier to maintain, but if it can be compiled to another lower level language like rust, it could improve [...]

I've wrote same Kalman filter in rust now. Here is the code
https://github.com/bareboat-necessities/wave_height_from_IMU/tree/main/rust

Might be next step is to write it using https://crates.io/crates/embedded-kalman

- embedded-kalman does not rely on the standard library and can be run in a #![no_std] environment
- The filters do not require dynamic memory allocation, and can be run without an allocator
- Plays nicely with embedded platforms such as the microcontrollers running on the ARM Cortex-M architecture

mgrouch, 2022-11-14:
Here is now working (on pi) rust code for getting vertical acceleration from IMU 

https://github.com/bareboat-necessities/wave_height_from_IMU/tree/main/imu

mgrouch, 2022-11-22:
I think I've come up with another way to calculate wave height and length from IMU and other basic sensors.
Without need to integrate acceleration. Acceleration noise is a bit difficult to deal with so I've added low pass filter before
using it.

A new method (without integration) is based on max/min acceleration and assumption that wave follows trochoidal model.
So now I can basically have 3 measurements
one on max acceleration with trochoid wave model
one on min acceleration  with trochoid wave model
and Kalman based integrator with drift prevention based on zeroing third integral (will be used with low freq low pass filter,
Butterworth or something)

The methods will take into account moving boat when calculating period and frequency using Doppler formulas.

More details can be found here: https://github.com/bareboat-necessities/wave_height_from_IMU/blob/main/trochoidal_wave.py
In comments and code.
Whole code (needs some more work) is here: https://github.com/bareboat-necessities/wave_height_from_IMU/

mgrouch, 2022-11-23:
The proposed algorithm also calculates 
true wind angle, true wind speed, speed through the water based on difference between GPS COG/SOG and true heading with apparent wind angle/speed. I think it might be useful for autopilot. 

Wave period (apart from wave height) is also calculated from just apparent wind parameters, SOG/COG, true heading and accelerometer

No need for speed over the water sensor, or other data.

mgrouch, 2022-11-27:
I've wrote small article to describe the algorithms I'm thinking on implementing.
Reviews are appreciated.

Here is the link https://bareboat-necessities.github.io/my-bareboat/bareboat-math.html

Thanks

rgleason, 2022-11-27:
Mike, thank you for your documentation, that's very helpful and interesting, but I am not the one who should comment.

seandepagnier, 2022-12-01:
> I've wrote small article to describe the algorithms I'm thinking on implementing. Reviews are appreciated.
> 
> Here is the link https://bareboat-necessities.github.io/my-bareboat/bareboat-math.html
> 
> Thanks

I will take a look at this when I get a chance...   the last few days I am navigating.

mgrouch, 2024-05-28:
I have found a way to estimate wave frequency from accelerometer without FFT, by using a simple on-line filter
(which also has low-pass component). 

I've read this paper and implemented it in python so far, with results exactly matching the article.

The algorithm is very lightweight and easy to run even on micro-controllers.

Aranovskiy on-line frequency estimator

Ref:

Alexey A. Bobtsov, Nikolay A. Nikolaev, Olga V. Slita, Alexander S. Borgul, Stanislav V. Aranovskiy

The New Algorithm of Sinusoidal Signal Frequency Estimation.

11th IFAC International Workshop on
Adaptation and Learning in Control and Signal Processing
July 3-5, 2013. Caen, France

https://www.sciencedirect.com/science/article/pii/S1474667016329421[The New Algorithm of Sinusoidal Signal Frequency Estimation]

Here is my code:
````

from math import sin, sqrt
import matplotlib.pyplot as plt
import numpy as np

# initialize algorithm constants

omega_up = 0.0  # upper frequency
a = 1.0
b = 1.0
k = 1.0
theta_0 = - omega_up * omega_up / 4.0
x1_0 = 0.0
sigma_0 = theta_0

delta_t = 0.001  # time step
count = 30000  # iterations

def func_y(t):
    return 2.0 * sin(2.0 * t)

# initialize variables
t = 0.0
x1 = x1_0
theta = theta_0
sigma = sigma_0

chart_x = np.zeros(count)
chart_y = np.zeros(count)

# loop
for n in range(0, count):
    # step
    y = func_y(t)
    x1_dot = - a * x1 + b * y
    sigma_dot = - k * x1 * x1 * theta - k * a * x1 * x1_dot - k * b * x1_dot * y
    theta = sigma + k * b * x1 * y
    omega = sqrt(abs(theta))

    print(omega)
    chart_x[n] = t
    chart_y[n] = omega

    x1 = x1 + x1_dot * delta_t
    sigma = sigma + sigma_dot * delta_t
    t = t + delta_t

plt.plot(chart_x, chart_y)
plt.grid()
plt.show()

````

Adjusted for Doppler effect from moving boat it could be a good estimator of wave frequency.
Wave directions in first approximation could be assumed following wind direction.

Now knowing just wave frequency and max vertical acceleration and making assumption that wave is trochoidal
it's simple to estimate wave height (using just algebraic formula) from my notes:

https://bareboat-necessities.github.io/my-bareboat/bareboat-math.html

Alternatively there is also a Kalman filter code in same notes which double integrates acceleration with built-in assumption
that on average boat stays afloat (for Kalman filter to converge). Which will give another wave height estimate which could be compared to the estimates from trochoidal model.

I've also found a way to detect waves directions from 3 or more accelerometers in another paper, but I didn't get to coding it.

Measuring wave direction can be achieved by using multiple accelerometers mounted in different locations on a ship.

Ref:

Ocean Engineering Volume 249, 1 April 2022, 110760

Johann A. Dirdala, Roger Skjetneb, Jan Roháčc, Thor I. Fossen

Online wave direction and wave number estimation from surface vessel motions using distributed inertial measurement arrays and phase-time-path-differences

https://www.sciencedirect.com/science/article/pii/S0029801822002098

mgrouch, 2024-09-01:
I have implemented wave height and heave estimation with mpu6886 IMU on esp32. 

The code is here:

https://github.com/bareboat-necessities/bbn-wave-period-esp32

My tests with attaching IMU to a broom stick and moving it up and down show good results. 

Estimating Wave Height using IMU

Estimate vessel heave (vertical displacement) in ocean waves using IMU on esp32

The method for estimating wave height and heave from a moving boat implemented here is using the following algorithm:

1. Estimate observed wave frequency using vertical boat acceleration measurements from IMU with Aranovskiy on-line filter. A frequency of acceleration sampling is important. This code seems able to achieve about 250Hz acceleration sampling from MPU6886. Lower frequencies wouldn’t give reasonable results from Aranovskiy filter.
1. Assume the waves follow trochoidal model. Trochoidal wave would still be observed as trochoidal from a vessel moving with constant speed. I.e. vertical movement is harmonic over time
1. Calculate observed wave period and calculate wave length using trochoidal wave model.
In trochoidal model horizontal wave displacement would be simply proportional to measured vertical acceleration. Doppler effect has no impact on displacement (amplitude). Although though it does on observed wave frequency and vertical acceleration in it.
1. Calculate the coefficient for previous step using known wave length in trochoidal model.
This gives a simple formula to produce vertical displacement from measured vertical acceleration. Valid in the considered models even by observations from a moving vessel (assuming speed doesn’t change much)
1. So this method avoids double integration of acceleration. It approximates a wave with trochoidal wave parameterized by two variables (frequency and max vertical acceleration in it) by observations from IMU on a moving boat. There is another way to use Kalman filter (with drift correction) which I might add later. See: https://bareboat-necessities.github.io/my-bareboat/bareboat-math.html However due to high accelerometer noise and low sample frequency I do not have high hopes for that Kalman filter.

Implementation Notes

- Tilt compensation. No magnetometer needed as we only need vertical projection of acceleation. Mahony algorithm using accel and gyro (without mag) is enough. Quaternion rotation is done for esimating vertical acceleration
- Rolling min/max algorithm with window of about three wave periods samples to produce wave height measurement. Algorithm: https://github.com/lemire/runningmaxmin from Daniel Lemire paper, and improvements from: https://github.com/EvanBalster/STL_mono_wedge

TODO

- Try to find a way for mpu6886 to sample at higher (than 250Hz) frequency

- Implement Kalman alorithm for wave estimation (the one mentioned here: https://bareboat-necessities.github.io/my-bareboat/bareboat-math.html) And compare with trochoidal model results.

- Generate some reference data, test cases with it and produce some graphical charts. There are many parameterized filters used and gains and coefficients need to be fine tuned for better results.

- Produce NMEA 0183 XDR sentences for heave, wave height, etc to serial port of m5atomS3

mgrouch, 2024-09-29:
A version of the sensor ready for sea trials is available now. 

https://github.com/bareboat-necessities/bbn-wave-period-esp32

Thanks!

mgrouch, 2025-05-10:
A release 2.0.5 is much improved version.

https://github.com/bareboat-necessities/bbn-wave-period-esp32

Thanks and fair winds!

mgrouch, 2025-07-15:
> This is awesome, Mike. How can this now be used to improve Pypilot or other autopilot software? Can it be modified to run without the esp32s3 using other precision devices such as [B&G Precision 9](https://defender.com/en_us/b-g-precision-9-compass-000-12607-001) which has a solid-state high-precision array sensor array based on MEMS technology which measures motion on nine separate axes: heading, heel, trim, roll, pitch, yaw, surge and sway?

It just needs IMU data to run. Sample rate should be around 200Hz for acceleration and less for magnetometer.
It could run on Linux or Windows too.