pypilot/pypilot #172: Calibration quality and local interference

orca-hydromancer, 2022-10-14:
I'm trying to figure out if there are any metrics/indicators from boatimu or calibration_fit to understand quality of the calibration.

Looking at [Precision 9](https://www.simrad-yachting.com/simrad/type/compasses/precision-9-compass/) compass [installation guide](https://www.defender.com/pdf/261066.pdf) I found the following

> Calibration
> The compass calibrates itself automatically while the boat is turning in a circle. However, the
> automatic calibration process must be triggered by user action. During the calibration, **the
> compass will measure the magnitude and direction of the local magnetic field. If the local
> magnetic field is stronger than the earth’s magnetic field (the local field is reading more than
> 100%), the compass calibration will fail. If the local field is reading more than 30%, you should
> look for any interfering magnetic objects and remove them, or you should move the compass
> to a different location. The local field angle will guide you to the local interfering magnetic
> object.**

https://www.thehulltruth.com/13476295-post4.html

> ...Field Angle and Field Strength data points.
> 
> This will indicate what relative direction ( with directly forward= 0*) from the P-9 compass/heading sensor you could find any strong offending onboard influence , and the Strength shows its influence relative to that of the Earth's local magnetic field.
> The Strength of any local onboard influence at the mounted position needs to be a very low % of that of Earth's magnetic field .

How it looks like in practice:
[image]
[credit](https://nakedsailor.blog/2020/05/17/calibrating-heading/)

Can pypilot provide local magnetic field magnitude and direction, or otherwise provide indication of local interference? The goal is provide some indication of reliability and quality of calibration.

seandepagnier, 2022-12-27:
You mean inclination angle and field strength?   These are both used for calibration but not much else.

The compass could work reasonably well with an old calibration even if the inclination and field strength have changed as long as they both remain fairly consistent,   so is this your intention for detecting magnetic distortions?

orca-hydromancer, 2022-12-27:
It might be. I want to use those values to indicate 1) if calibration might be bad due to magnetic interference and 2) direction of that interference. For me calibration is not running continuously, I disable it after a decent fit, but I want to have those values during calibration and at the end to be able to determine if calibration needs to be done again and if there is potential interference that should be removed before re-doing it.

Re [field strength](https://github.com/pypilot/pypilot/blob/0f62d9f73f5221710d0d1cd14d1a6e1258ec627f/pypilot/calibration_fit.py#L541) what do you think about suggestion from Simrad: if the field strength >100% calibration will fail and if >30% look for and remove interference? I see pypilot code is discarding calibration when field strength <7 or >120, which seems pretty different range compared to Simrad (if the values are comparable).

Re [inclination](https://github.com/pypilot/pypilot/blob/0f62d9f73f5221710d0d1cd14d1a6e1258ec627f/pypilot/calibration_fit.py#L548) how can it be used to determine the direction of interference (as suggested by Simrad)? I see pypilot is discarding calibration when inclination > 82, but my understanding is that (in case of Precision-9) field angle is relative to compass forward (0), and should be only used as indication of direction of interference conditional on field strength indicating strong interference (>30%).

Since I'm disabling calibration after a decent fit (due to CPU limits), I'm unable to detect changes in the field strength/angle. Fine for now, but in the future I'll look into checking these values periodically to understand if I need to re-calibrate.

seandepagnier, 2022-12-28:
Basically,   it needs to verify that the field strength and/or inclination varies with reading somehow to indicate the bias is actually incorrect.

There is no way to know the direction of the interference for sure because multiple sources of magnetic interference are possible.    With a single source, you would have to basically perform a calibration and compare it to the previous calibration.

I dont know what simrad does because there is no source code.    The range in pypilot covers the range of magnetic strength anywhere on earth.    It is true that very near the magnetic poles  pypilot cannot calibrate the compass either but to my knowledge no one has used pypilot there, and it would still steer, just not able to recalibrate.

You can simply compute the magnetic field strength by applying compass bias and scale and computing the magnitude and use the dot product to compute inclination.     I suppose I could create  imu.compass.inclination and imu.compass.field_strength or something, that basically would not normally be accessed, and could be computed only if someone is watching them...