pypilot/pypilot #173: Magnetic deviation in boatimu

orca-hydromancer, 2022-10-14:
Does pypilot provide magnetic deviation for heading as described in [NMEA HDG msg](https://gpsd.gitlab.io/gpsd/NMEA.html#_hdg_heading_deviation_variation) ?

Magnetic variation can be obtained from [WMM2020](https://www.ngdc.noaa.gov/geomag/WMM/). What about magnetic deviation?

I see that calibration_fit is calculating some kind of deviation in ComputeDeviation, wondering what the unit is, and how it relates to deviation value as defined in HDG.

seandepagnier, 2022-12-27:
To compute magnetic deviation needs gps.    It does actually compute this with an additional kalman filter which is generally disabled:
https://github.com/pypilot/pypilot/blob/master/pypilot/gps_filter.py

Perhaps instead of magnetic deviation I have called it "compass offset"  and instead of magnetic variation I used the term inclination,  but it could be good to rename these.

the deviation in calibration_fit is to ensure quality of measurements so  generally it will fail if the sensors have errors, or there are many magnetic distortions.

orca-hydromancer, 2022-12-27:
Magnetic [declination is same as variation](https://www.ngdc.noaa.gov/geomag/declination.shtml), for which GPS is needed. I have this via [WMM2020](https://www.ngdc.noaa.gov/geomag/WMM/), same as [gps_filter.py](https://github.com/pypilot/pypilot/blob/master/pypilot/gps_filter.py)

[Magnetic deviation](https://en.wikipedia.org/wiki/Magnetic_deviation) seems to an offset that needs applying to get actual magnetic heading and it seems to be pypilot's [compass_offset](https://github.com/pypilot/pypilot/blob/a7b4a1c53f6ad39081295c4f5d819ab232d417f5/pypilot/gps_filter.py#L351). As far as I can tell this is computed automatically, but not sure from what data/sensor and if it can be fit within boatimu, can you enlighten?

I'm trying to produce HDG message, and currently providing magnetic heading and magnetic variation, but wondering what I should put for magnetic deviation.

seandepagnier, 2022-12-28:
It is a tough one.    There is a difference of which direction the boat is facing vs the direction it is moving which is leeway.   This is combined with compass alignment errors as well as magnetic distortions caused by metal on the boat.

So basically the intention is to have a button you can press when the boat is sailing fast enough without any leeway (running  directly downwind maybe)   and there are no currents or local magnetic distortions.     Assuming the declination or variation is known  pypilot supports WMM2020,  it could compute a compass_offset and update the heading alignment.

Without this button, it can actually automatically compute the error over time using the accelerometers (in theory) regardless of leeway currents or magnetic distortions  which is what is implemented but not really proven.

Once this is done,   if there are currents, leeway or local magnetic distortions (not on the boat which are compensated at this point hopefully, but like wrecks or going under steel bridges), it may be impossible to distinguish between, but it would be possible to measure the combination of these effects and call this  magnetic variation?   It would also include errors in the magnetic model and so forth.

Finally...  some boats get re magnetized if docked facing north long enough,  or along a passage,  and certainly as boats cross oceans the magnetic fields change significantly.    Pypilot does not really compensate for these cases, but the calibration is formulated so that it should still be usable if calibrated from a different geographical location.