pypilot/pypilot #174: Hat and new designs

mgrouch, 2022-10-20:
Is it possible for new designs to move away from using rpi hat and put that functionality (display and remote)
on motor controller MCU?

RPI Hat makes design very locked in into Raspberry Pi vendor.

I'd like to consider possibilities of running PyPilot on other boards. Like Rock Pi, etc under Armbian.

Thanks

PeterBro3, 2022-10-20:
In my installation the TinyPilot (or PyPilot) is at the nav station.  The motor controller(s) are in an aft lazarette close to the autopilot hydraulics and quite inaccessible.  It wouldn't work to have display or remote in that location.  Aren't you really asking about also having hat equivalent hardware for other autopilot microcontrollers?

mgrouch, 2022-10-20:
>  It wouldn't work to have display or remote in that location.

So you can change location with longer wire.

PeterBro3, 2022-10-20:
30 ft of wire carrying 25 amps from motor controller to motor (x2 because I have 2 motor/motor controllers) not a good idea!  It's an unusual installation but the distance and size of motors is common for larger boats.

seandepagnier, 2022-10-20:
it would be possible but there are 2 issues:

1)  the mcu used by the motor controller has no more spare gpio pins
2)  it would go against my design philosophy

Of course everything being free software someone may hack this feature in.   Someone added an lcd display to the motor controller, but I did not really give glowing reviews as it distracted the processor from its primary function.

What I would suggest is,  using a pypilot hat on a different processor.   The 40 pin gpio is becoming something of a standard, and orange pi already did work a few years ago.    If needed, a different hat or adapter board could be developed.    I have development of wireless keypads and wireless remotes already work.   This gives the most flexible option and it is the path I will proceed in the future.    So even if wires are leading to the motor,  a wireless remote could be attached there as well.   Wired buttons are cumbersome already because of needing a wire for each button.   A protocol with fewer wires may be possible but wireless already works.    I had also the idea to potentially "tap" the power to the motor with a rectifier to reduce drain from any battery, but in practice 2xAAA batteries will last for years.

closed 2022-10-24 by mgrouch