pypilot/pypilot #177: Like to use external Compass for heading, pitch and roll, not depending on IMU.

hans-rickardt, 2022-10-24:
Have an Autohelm 3000 that I plan to convert to PyPilot.
The sailboat have Raymarin plotter axiom 9 ,  on the NMEA2000/Seatalk^NG bus I have GPS, Wind, logg, rodder angle and an Ardruino configure with and Compass BNO055 and NMEA2000 that provide Heading, yaw, pitch, roll. On the NMEA2000 bus I also have a Raspberry PI running SignalK, OpenCPN, 4G USB as WiFi-AP.

I noticed that PyPilot Sends zero for yaw, pitch and roll even if no IMU is connected to the Raspberry PI.
Like the PyPilot “nmea.py” updates so that I can use an external source for yaw, pitch, roll  via SignalK.

Hans

seandepagnier, 2022-10-24:
gyros or rate of change is most important.   Furthermore,  using external sensors here tend to introduce lag and errors so I generally discourage it.   The performance will be worse and I fear that if this were common and results shared, it could show pypilot working poorly.

hans-rickardt, 2022-10-24:
Yes I can agree some over head will it be but the NMEA2000 bus speed is 250k bps and some protocol over head so we talking about 2-10ms for a data frame. An external rudder sensor is working today in my test rig. 
But it is not a big problem for me to install an additional IMU

Next step in my project will be to write a module that simulate a Raymarine p70 Autopilot :-) so that I can control PyPilot from Axiom

mgrouch, 2022-10-25:
In aviation drone autopilots run on lower bandwidth and speed than NMEA 2000.
Autopilot is a part of RC controls of a drone. Or am I wrong with my statements?

10ms is way too high for NMEA 2000. My round trip ping to google from my house is 7ms.
CPU instructions are in nano seconds. Between nano and milli there is micro.
250Kbs speed would be in micro not milli seconds.

 ```
ping www.google.com

Pinging www.google.com [142.250.65.228] with 32 bytes of data:
Reply from 142.250.65.228: bytes=32 time=7ms TTL=119
Reply from 142.250.65.228: bytes=32 time=8ms TTL=119
Reply from 142.250.65.228: bytes=32 time=7ms TTL=119
Reply from 142.250.65.228: bytes=32 time=7ms TTL=119

Ping statistics for 142.250.65.228:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 7ms, Maximum = 8ms, Average = 7ms
```

Thanks

PS: I think I've replied to wrong thread in openplotter. This is where I aimed it at