pypilot/pypilot #178: Command line utilities for emulating remote control button actions

mgrouch, 2022-10-31:
I'd like to try implementing a service for using bluetooth devices as remote control.
I can capture BT buttons codes using 

https://github.com/bareboat-necessities/btbutton (might need some more work)

If I pipe it to shell script I could invoke command line tools to perform remote control button actions

What is the way to invoke those actions from a shell script?

Thanks,

mgrouch, 2022-11-02:
Or might be use this

https://github.com/wertarbyte/triggerhappy

seandepagnier, 2022-11-02:
my biggest concern for manual control is lag and timing.   So pressing and holding the button I dont know how fast the updates arrive.  It is basically difficult to work on this solution without the actual remote, but I would add it to pypilot_hat and allow these buttons to be programmed and assigned similarly to all the other buttons.

Invoking pypilot_client can perform many pypilot actions but will increase lag as well as not be able to interact with the lcd support.