pypilot/pypilot #179: Will pypilot honor waypoint sentences from OpenCPN?

mgrouch, 2022-11-01:
Let's say I will create server TCP NMEA 0183 connection in OpenCPN. 
I will make OpenCPN to transmit autopilot waypoint related data sentences.
I see when I activate a route it transmits some sentences with waypoint data using EC talker ID.

Then I will create connection in pypilot to that NMEA 0183 port.

Will pypilot honor active route set in OpenCPN and navigate to next active OpenCPN waypoint?

Thanks

Rudder2, 2023-04-14:
Yes, with great accuracy I have to say also.  I have a BoatIO Hat that has a Ublox GNSS GPS Chip on it attached to my OpenPlotter Raspberry PI4 with OpenCPN transmitting waypoints to PyPilot it can navigate my boat dead center of a canal between bays.  I'm impressed.

seandepagnier, 2023-04-14:
This is really not an issue but a question.

@mgrouch could you place inquiries on the forum in the future?   it makes it easier to track actual issues.

closed 2023-04-14 by seandepagnier