pypilot/pypilot #186: Improper use of the IMU regarding rate

Asw1n, 2023-01-27:
the program boatimu. py calculates rate of turn (over three axes) from the gyro data starting at line 417. The gyro data is first rotated using the rotation vector from the IMU and then a low pass filter is applied to get headingrate_lowpass.

1) I think however that this is not the way an IMU should be used. The heading rate should be calculated from the change in IMU state, like rate = (fusionQPose_at_T - fusionQPose_at_t-1) /  dt. The reason for this is that the Kalman filter gives the best estimate of orientation, and therefore the derivative of the Kalman filter pose gives the best estimation of rate. This is a better estimation than  the measurement from the gyro sensor itself. One reason for this is that the Kalman filter has removed the sensor noise from the gyro in its pose. This also means that when using the pose from the filter there is no need to apply a low pass filter on the rate data.

2) When using the current method and also when using the proposed method the rate data should be be aligned as well. But it is not.   Alignment takes place after rate is calculated.

Asw1n, 2023-01-28:
An adddition/correction.
Rate probably is part of the filter state, so it might be queried from the filter. Still not being the gyro redding.

seandepagnier, 2023-01-29:
First of all,   it would gain nothing useful from this if the kalman filter has only gyro, compass and accel.

The accelerometer has noise from lateral acceleration and the compass from magnetic distortions,   furthermore both of these can be uncalibrated.    So you would get a worse result.     Finally...   the kalman filter is fusing gyro and other sensors in a way to eliminate gyro drift and integration error but it wont improve the rate of change data without additional sensors, but in that case it could make sense .   For example a video camera that can see the horizon.

Now feel free to try, but you will just be adding more noise and potentially have a very bad result (uncalibrated) in the best possible case there will be no useful improvement.

Asw1n, 2023-01-29:
The Kalman filter does improve rate estimation when it only has gyro, compass and accel. It improves rate estimation over the Z-axis with the compass and it improves rate estimation over the X and Y-axis using the accelerometer. 
That is even the case when there is noise from other sensors. Because even when you do not take rate from the filter but from the gyro sensor you will introduce the noise and disturbances from the compass and accelerometer into your estimation as you have to rotate the gyro data to world frame using the (noisy or disturbed) pose that is estimated by the filter. The same is true when these sensors are not calibrated. Also the system as it is will suffer from this as the gyro data needs to be rotated using the estimation of the Kalman filter.

The thing is that the filter does not add noise, it reduces noise, also in the rate estimation. Of course all the sensors need to be calibrated for the filter to work optimally, but that is also true in the current implementation.

Let me finish with mentioning two drawbacks of the current implementation where the proposed implementation will perform better. The current implementation uses a low pass filter to filter out the high frequency noise of the gyro. This introduces a lag in the rate estimation. Also the low pass value (≈ the time constant) of the low pass filter is a more or less arbitrary guess. Guesses in the Kalman filter are based on a sound theoretical framework and on known sensor characteristics.

I will try to test my proposal myself, but I need to learn how to write python and how to create an installable image of pypilot first.