pypilot/pypilot #189: rudder position for instruments via NMEA

mgrouch, 2023-03-21:
Return rudder position for instruments via NMEA

Thanks

rgleason, 2023-03-21:
Mgrouch. I don't believe you are referring to pypilot_pi, but there is a divergence issue between sean and my watchdog repos involving the rudder alarm and the pypilot alarms. I've been working on it for awhile locally. I have a version that works with rudder alarms and no pypilot_pi alarms and another version that is the reverse. Hopefully I can recreate the rudder alarms in sean's version at some point.

mgrouch, 2023-03-21:
Not talking about alarm or plugin. 
Rudder position is currently not reported by pypilot in its NMEA stream. ROT is reported. But not rudder position (RSA rudder angle sentence)

thanks