pypilot/pypilot #201: Firware

Max194703, 2023-04-20:
It is possible to create a firmware update function on the web?

seandepagnier, 2023-04-21:
Which firmware?    I recently implemented support to update the firmware on the microprocessor on the hat.  It will happen automatically on newer boards.

Or do you mean for pypilot itself on tinypilot?   It could work but only if the pilot is a client with internet access.   Or do you mean, to browse for a file you download and use that?

Max194703, 2023-04-24:
I mean to be able to upload updates from the web interface, see: https://pcnautic.gitbook.io/autopilot/bediening/webinterface/configuration-configuratie/firmware-upgrade

seandepagnier, 2023-04-25:
I have not merged this feature into my repository yet but I will consider it for the future.

Max194703, 2023-04-26:
Ok, tks, buon vento!

closed 2023-11-07 by seandepagnier