pypilot/pypilot #210: PID controller vs fractional order controller

mgrouch, 2023-10-13:
Hi Sean,

Have you considered using fractional order controller instead of current basic PID controller? According to this article they offer some benefits. 

Linear fractional order controllers; A survey in the frequency domain

Ali Ahmadi Dastjerdi a, Blas M Vinagre b, YangQ

https://www.sciencedirect.com/science/article/abs/pii/S136757881830172X

Instead of first order integral and derivative they use fractional integral / derivative

Some introduction on fractional calculus:

https://youtu.be/A4sTAKN6yFA?feature=shared

Very interesting stuff. 

Thanks,
—MG

mgrouch, 2023-10-13:
Fractional-Order Controller for Course-Keeping of Underactuated Surface Vessels Based on Frequency Domain Specification and Improved Particle Swarm Optimization Algorithm

by [Guangyu Li](https://sciprofiles.com/profile/2121724?utm_source=mdpi.com&utm_medium=website&utm_campaign=avatar_name) 1ORCID, [Yanxin Li](https://sciprofiles.com/profile/author/ZllaQUNDQklOYUFMTGhpbzRRQU8zKytXMW1FT0NpNEltVW04R2wwUHZTVT0=?utm_source=mdpi.com&utm_medium=website&utm_campaign=avatar_name) 1, [Huayue Chen](https://sciprofiles.com/profile/1094446?utm_source=mdpi.com&utm_medium=website&utm_campaign=avatar_name) 2,* and [Wu Deng](https://sciprofiles.com/profile/300277?utm_source=mdpi.com&utm_medium=website&utm_campaign=avatar_name) 3,*

https://www.mdpi.com/2076-3417/12/6/3139

seandepagnier, 2023-10-22:
this is really very interesting, and I have learned how to take fractional-order derivatives however, I am not quite sure how to implement this in pypilot.     How would I take heading error, and rate of change and produce a stream of fractional order?

mgrouch, 2023-10-22:
Have you looked at second article in links

https://www.mdpi.com/2076-3417/12/6/3139

?

Thanks

mgrouch, 2023-10-22:
One implementation 

https://github.com/extall/fopid-controller-prototype

mgrouch, 2023-10-22:
> this is really very interesting, and I have learned how to take fractional-order derivatives however, I am not quite sure how to implement this in pypilot. How would I take heading error, and rate of change and produce a stream of fractional order?

Just like in PID controller. Only instead of first order integral and first order derivative you use parameterized order integral and parameterized order of derivative. 

This adds two more parameters in controller.
5 in FOPID vs 3 in classic PID. 

Autotuning (finding near optimal) 5 parameters is performed (in the paper) using particle swarm algorithm. 

Numeric calculations of fractional calculus operators is done using improved Oustaloup approximation for the FO calculus. 

The derivative and integral they use in paper are Caputo.

rgleason, 2023-10-23:
Since Sean is a math wiz he probably already knows that AI has found faster ways to do matrix calcs.  Good for smaller processors?

mgrouch, 2023-10-23:
> Since Sean is a math wiz he probably already knows that AI has found faster ways to do matrix calcs. Good for smaller processors?

I haven’t seen AI making discoveries. It usually just digs via known facts. Can you share some paper on what you mentioned? How do you think it’s relevant to autopilot?

rgleason, 2023-10-23:
MIT technology wrote about it and it was all over the press.  I posted it in CF OpenCPN and they removed it as "Off-topic"
Just google [known facts](https://nortonsafe.search.ask.com/web?annot=false&vendorConfigured=ask&o=APN12179&prt=SSS&ver=3.21.0.6&tpr=111&chn=store&guid=a2ea09b4-77cd-4198-9d3c-d6faddf528f0&doi=2023-01-27&q=AI+improves+matrix+calculations&geo=en_US&ctype=&darkMode=false&sameTabLaunch=false&browser=FireFox&prod=HP&installSource=other&cmpgn=medici)

https://www.quantamagazine.org/ai-reveals-new-possibilities-in-matrix-multiplication-20221123/
https://arstechnica.com/information-technology/2022/10/deepmind-breaks-50-year-math-record-using-ai-new-record-falls-a-week-later/

https://arstechnica.com/information-technology/2022/10/deepmind-breaks-50-year-math-record-using-ai-new-record-falls-a-week-later/

https://thenewstack.io/how-deepminds-alphatensor-ai-devised-a-faster-matrix-multiplication/

seandepagnier, 2023-10-23:
rick,  it is  offtopic even here.

anyway yes AI can solve certain specific problems like this.

As for the extended pid, I understand 2 added parameters.  I am not familiar with the oustaloup method.   If particle swarm works for autotuning this, it would work for a basic PID filter as well and in my mind that should be working first.

mgrouch, 2023-10-23:
> rick, it is offtopic even here.
>
> anyway yes AI can solve certain specific problems like this.
>
> As for the extended pid, I understand 2 added parameters. I am not familiar with the oustaloup method. [...]

A library in python for particle swarm with examples 

https://github.com/guofei9987/scikit-opt#3-psoparticle-swarm-optimization

Thanks

mgrouch, 2023-10-23:
> A library in python for particle swarm with examples
>
> https://github.com/guofei9987/scikit-opt#3-psoparticle-swarm-optimization
>
> Thanks

Particle swarm optimization (PSO) in Julia programming language 

https://julianlsolvers.github.io/Optim.jl/stable/#algo/particle_swarm/

rgleason, 2023-10-23:
Ok but aren't matrix a necessary part of the math for this idea?

mgrouch, 2024-01-17:
I see number of python to Julia transpilers. Both languages are similar. Julia programs are compiled and run faster though.
python is a scripting language and once you do production systems in it there are many dependencies management issues.
With Julia it's possible to deliver the software as compiled binaries.
I'm just brainstorming but might be switching from python to Julia is a valid idea.

Thanks

seandepagnier, 2024-01-17:
There is no issue with calling other languages for specific portions of pypilot if much computation is needed and speed is an issue.   Already pypilot calls c or c++ for a lot of things to reduce cpu usage.    It even invokes fortran for the linear algebra portions.    It would be similarly possible to call julia or rust or any other language from python.   It is also possible to compile python.

I have choosen python for readability and ease of modifications.  So far it is demonstrated to be sufficient in performance for the current algorithms.   Unless there is a demonstrated reason to switch to say julia I do not understand why this would be helpful and it would make the program more difficult to maintain.   <3% cpu on raspberry pi is used (more for pi zero but less than half for all processes)      Had I written pypilot entirely in c++ for example, it would be years behind its current state.   Look at opencpn: it has taken away thousands of hours tracking  bugs that only occur because of that unsafe language.

while julia may one day supersede python and at that point a switch could be considered: it has not at this point in my mind.   It is more important to use a language that more people are familiar with than to save 1% of processing power.