pypilot/pypilot #213: pypilot_web won't start; import error

marcobergman, 2023-11-16:
Can't get it to start now, if throws `ImportError: cannot import name 'Markup' from 'flask' (/usr/local/lib/python3.9/dist-packages/flask/__init__.py)`

Pretty straight from github; see attached.
[pypilot_web-import-error.txt](https://github.com/pypilot/pypilot/files/13385167/pypilot_web-import-error.txt)

seandepagnier, 2023-11-17:
you must install the required dependencies, perhaps yours are a too recent version?

marcobergman, 2023-11-17:
I thought the dependencies were (quite brilliantly) resolved by setup.py now, and expected the solution to this issue would lie in there.

seandepagnier, 2023-11-17:
it is supposed to, but it seems you are using python 3.9 which newer than what I have ever used (3.8) and perhaps some of the dependencies of pypilot have changed in their latest versions which break things.

So technically pypilot needs to install maybe a specific version of flask?   For example 1.0.2?

What do you have for __version__ = /usr/local/lib/python3.9/dist-packages/flask/__init__.py

Could you try "pip install flask==1.0.2"  ?

rgleason, 2023-11-17:
Sean, thanks for finding that. In order to build all the OS, we need to keep pip up to a certain version so that CircleCI can build successfully. It is a moving target as code goes out of date and is deprecated. We will need to figure out a working solution to this for all the builds.

marcobergman, 2023-11-18:
Don't know why I have this python; I must have done apt upgrade.

> pip install flask==1.0.2

I did this, got a message, then did pip install flask==2.0, that got rid of the `ImportError: cannot import name 'Markup' from 'flask'`. 

Now I have a new one: `ImportError: cannot import name 'url_quote' from 'werkzeug.urls'`.

```
pi@openplotter:~ $ pypilot_web
[...]
ImportError: cannot import name 'Markup' from 'flask' (/usr/local/lib/python3.9/dist-packages/flask/__init__.py)

pi@openplotter:~ $ pip install flask==1.0.2
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
flask-babel 4.0.0 requires Flask>=2.0, but you have flask 1.0.2 which is incompatible.

pi@openplotter:~ $ pip install flask==2.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting flask==2.0
  Downloading https://www.piwheels.org/simple/flask/Flask-2.0.0-py3-none-any.whl (93 kB)
     |████████████████████████████████| 93 kB 435 kB/s
Requirement already satisfied: Werkzeug>=2.0 in /usr/local/lib/python3.9/dist-packages (from flask==2.0) (3.0.0)
Requirement already satisfied: itsdangerous>=2.0 in /usr/local/lib/python3.9/dist-packages (from flask==2.0) (2.1.2)
Requirement already satisfied: click>=7.1.2 in /usr/local/lib/python3.9/dist-packages (from flask==2.0) (8.1.7)
Requirement already satisfied: Jinja2>=3.0 in /usr/local/lib/python3.9/dist-packages (from flask==2.0) (3.1.2)
Requirement already satisfied: MarkupSafe>=2.0 in /usr/local/lib/python3.9/dist-packages (from Jinja2>=3.0->flask==2.0) (2.1.3)
Installing collected packages: flask
  Attempting uninstall: flask
    Found existing installation: Flask 1.0.2
    Uninstalling Flask-1.0.2:
      Successfully uninstalled Flask-1.0.2
  WARNING: The script flask is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed flask-2.0.0

pi@openplotter:~ $ pypilot_web
[...]
ImportError: cannot import name 'url_quote' from 'werkzeug.urls' (/usr/local/lib/python3.9/dist-packages/werkzeug/urls.py)

```

qMalte, 2024-05-29:
Hi Marco,

which Raspberry version are you currently using?
I have found that the PyPilot runs much better with the version based on Debian 11 (Bullseye), at least with regard to the installation script.

Best regards
Malte

marcobergman, 2025-02-11:
Just cloned a fresh pypilot and it compiles and starts pypilot_web without any problems, so issue closed. 

marcb@raspberry:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
marcb@raspberry:~ $ python --version
Python 3.9.2

closed 2025-02-11 by marcobergman