pypilot/pypilot #214: Remove dependency on wiringpi

mgrouch, 2023-12-16:
WiringPi is not working on pi5 and bookworm. 
wiringPi is no longer maintained and is deprecated 

thanks

seandepagnier, 2023-12-16:
wiringpi should be optional.   Are you getting an error message from it?

mgrouch, 2023-12-16:
I’ll try without wiring pi. Thanks

mgrouch, 2023-12-18:
It runs without wiringpi on bookworm. (So far tested on pi4, not - pi5). It gets heading from IMU fine.

However pypilot_calibration fails on start under bookworm:

```

user@lysmarine:~ $ pypilot_calibration 
Traceback (most recent call last):
  File "/usr/local/bin/pypilot_calibration", line 33, in <module>
    sys.exit(load_entry_point('pypilot==0.36', 'console_scripts', 'pypilot_calibration')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 426, in main
    glutInit(sys.argv)
  File "/usr/lib/python3/dist-packages/OpenGL/GLUT/special.py", line 333, in glutInit
    _base_glutInit( ctypes.byref(count), holder )
  File "/usr/lib/python3/dist-packages/OpenGL/platform/baseplatform.py", line 423, in __call__
    raise error.NullFunctionError(
OpenGL.error.NullFunctionError: Attempt to call an undefined function glutInit, check for bool(glutInit) before calling

```

seandepagnier, 2023-12-18:
could it be that glut or at least glut in python is broken?

seandepagnier, 2023-12-18:
Maybe it works now?

mgrouch, 2023-12-18:
It goes further but now fails with this:

```

user@lysmarine:~ $ pypilot_calibration 
Traceback (most recent call last):
  File "/usr/local/bin/pypilot_calibration", line 33, in <module>
    sys.exit(load_entry_point('pypilot==0.36', 'console_scripts', 'pypilot_calibration')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 429, in main
    CalibrationDialog().ShowModal()
    ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 30, in __init__
    super(CalibrationDialog, self).__init__(None)
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_control_ui.py", line 340, in __init__
    self.m_panel4.SetSizer( fgSizer22 )
wx._core.wxAssertionError: C++ assertion "CheckExpectedParentIs(w, m_containingWindow)" failed at ./src/common/sizer.cpp(887) in SetContainingWindow(): Windows managed by the sizer associated with the given window must have this window as parent, otherwise they will not be repositioned correctly.

Please use the window wxPanel@0x295c77e0 ("panel") with which this sizer is associated, as the parent when creating the window wxGLCanvas@0x29b750e0 ("GLCanvas") managed by it.
Exception ignored in: <function CalibrationDialogBase.__del__ at 0x7f77024680>
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_control_ui.py", line 880, in __del__
    self.m_notebook.Unbind( wx.EVT_NOTEBOOK_PAGE_CHANGED, None )
  File "/usr/lib/python3/dist-packages/wx/core.py", line 1493, in _EvtHandler_Unbind
    return event.Unbind(self, id, id2, handler)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/wx/core.py", line 1562, in Unbind
    success += int(target.Disconnect(id1, id2, et, handler))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: wrapped C/C++ object of type Notebook has been deleted

```

seandepagnier, 2023-12-18:
I am going to need to reproduce this on my end to fix I think...

mgrouch, 2023-12-18:
@seandepagnier 

easy way to reproduce is just 
burn latest test bookworm image of BBN OS
and load it on pi4. 

https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/

and just start pypilot_calibration from command line.