pypilot/pypilot #216: pypilot_calibration issues on bookworm

mgrouch, 2024-01-14:
````

user@lysmarine:~ $ pypilot_calibration 
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 150, in receive_messages
    self.enumerate_settings(values_list)
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 135, in enumerate_settings
    proc()
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 125, in proc
    s.SetDigits(-math.log(s.GetIncrement()) / math.log(10) + 1)
TypeError: SpinCtrlDouble.SetDigits(): argument 1 has unexpected type 'float'
Unimplemented OBJ format statement 's' on line 's off'
Unimplemented OBJ format statement 's' on line 's 1'
Unimplemented OBJ format statement 's' on line 's off'
Unimplemented OBJ format statement 's' on line 's 1'
Unimplemented OBJ format statement 's' on line 's off'
Unimplemented OBJ format statement 's' on line 's 1'
Unimplemented OBJ format statement 's' on line 's off'
Unimplemented OBJ format statement 's' on line 's 1'
Unimplemented OBJ format statement 's' on line 's off'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 53, in <lambda>
    self.Bind(wx.EVT_TIMER, lambda e : self.sHeadingOffset.SetValue(round3(self.pypilot_heading_offset)), id = self.ID_HEADING_OFFSET)
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: SpinCtrl.SetValue(): arguments did not match any overloaded call:
  overload 1: argument 1 has unexpected type 'float'
  overload 2: argument 1 has unexpected type 'float'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 154, in receive_messages
    self.receive_message(msg)
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 259, in receive_message
    self.UpdatedSpin(self.sRudderRange, value)
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 178, in UpdatedSpin
    self.UpdateControl(dspin, lambda : dspin.SetValue(value))
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 171, in UpdateControl
    update()
  File "/usr/local/lib/python3.11/dist-packages/pypilot/ui/autopilot_calibration.py", line 178, in <lambda>
    self.UpdateControl(dspin, lambda : dspin.SetValue(value))
                                       ^^^^^^^^^^^^^^^^^^^^^

````

Thanks

rgleason, 2024-01-16:
@mgrouch   Please try the most recent PIM version 0.33.40.0  which fixed android crash and implemented calibration for android. The changes may have fixed this problem.

mgrouch, 2024-02-08:
> @mgrouch Please try the most recent PIM version 0.40.33.0 which fixed android crash and implemented calibration for android. The changes may have fixed this problem.

This issue is not with pypilot plugin. It’s with pypilot own calibration in python. 
It seems it’s missing some round() calls to convert floats to integer when setting precision and values of UI spinners.

mgrouch, 2024-02-14:
Fixed in latest BBN OS Image.
I think it can be closed

closed 2024-02-14 by mgrouch