pypilot/pypilot #219: PypilotHat support on pi5

mgrouch, 2024-02-10:
Does it require pigpiod?
Anything gpio pi specific doesn’t work on pi5
so looks like it needs to be fixed somehow. 

More on it here
https://forums.raspberrypi.com/viewtopic.php?t=359742

Thanks

seandepagnier, 2024-02-10:
i need to get a pi 5

mgrouch, 2024-02-10:
In https://github.com/pypilot/pypilot/blob/master/setup.py

````ugfx_defs = ['-DWIRINGPI']
try:
    import RPi.GPIO
    ugfx_libraries=['wiringPi']
except:
    try:
        import OPi.GPIO
        ugfx_libraries=['wiringPi']
    except:
        print('no RPi.GPIO library for ugfx')
        ugfx_libraries=[]
        ugfx_defs = []
````

leads to error 'no RPi.GPIO library for ugfx'

mgrouch, 2024-02-11:
Related: https://github.com/bareboat-necessities/lysmarine_gen/issues/376