pypilot/pypilot #221: OpenCPN plugin fails to connect to pypilot if config dialog window not visited

mgrouch, 2024-02-12:
OpenCPN plugin fails to connect to pypilot if config dialog window not visited.

I have to open/close config dialog of pypilot plugin on each OpenCPN restart

seandepagnier, 2024-02-12:
try   pulling from git master I fixed a stupid bug just now

mgrouch, 2024-02-12:
Would it be in https://github.com/pypilot/pypilot_pi  ?

I do not see any changes

marcobergman, 2024-03-02:
Where was this change made? Don't see it either. Thx.

mgrouch, 2024-03-02:
> Where was this change made? Don't see it either. Thx.

There was no change to fix it. It’s still not fixed. 
I’ve reported it in OpenCPN  plugin github projects.

Steve111888, 2024-12-19:
"There are GPIO conflicts between the following apps: GPIO - 1W, pypilot - pypilot hat" ???