pypilot/pypilot #223: IMU redundancy

mgrouch, 2024-02-20:
I think IMU is a very critical piece. 

To avoid misreadings and drifts I think 
it could make sense to support three IMUs. 

If two ones give close values and another one is gives values significantly off then use only values
from two good ones. possibly averaged. 

just brainstorming…
Thanks

seandepagnier, 2024-02-21:
With 3 imu, they can all be placed such that no 2 axes are parallel.    By averaging the sensors, the noise level can be reduced to 1/sqrt(3)    This also gives the possibility of detecting sensor failure and continuing to work.

It is possible with only 2 imu as well, but  in the case where one imu gives slightly wrong readings on all of its sensors it is not possible to detect.

mgrouch, 2024-02-21:
Software and its documentation need to support it though. ;)

mgrouch, 2024-03-09:
What I found by observing pypilot heading readings from IMU sitting still at home over time that magnetometer is extremely sensitive. 

Just putting a hand next to it about 10 cm away changes readings by 10 degrees. So forget about calibrating it holding in your hand. Enclosure needs to be big. 

Closing laptop lid a foot apart in distance changed readings by 15 degrees. 

So I really started think that having one IMU is not a valid option for pypilot. 

Thanks

seandepagnier, 2024-03-10:
Is there something magnetic in your hand?

What do you mean one IMU is not valid?     It is true that putting magnetic objects near the compass will cause issues, but it does ok as long as you dont do this.