pypilot/pypilot #225: Implemented motor.iso sketch for Nano Every, do you want it?

DarfieldChris, 2024-03-27:
Hi,

Not sure where to post this, so I posted it in issues.

I planned to try out the pypilot autopilot some time ago, bought my parts and never did anything. This is a large project, very impressed with what has been done!

I am currently cruising in the Bahamas and my existing autopilot failed so over the last week have been seeing if I can get pypilot working.

I thought I bought an Arduino Nano for this project but in reality was sold a Nano Every (I think the assumption is that it is pin compatible so why not use an Every in place of the old Nano).

As I am guessing you know (and I painfully found out!) the Nano Every is not pin compatible, nor is it programmed the same as the older Nano as it uses a different processor.

Regardless, I ported your existing motor.iso to the Nano Every (see the attached zip). I used ifdef statements for the new board so it should work for either board now though I have only been able to test it on the Every, so not sure if I broke anything from the original implementation but I do not think I did. 

I would be happy to remove the #ifdef statements and just provide a motorNanoEvery.iso file if you want to include it in your repository as the code as it exists is now very hard to follow.

From the modified code:
/* Ported to Arduino Nano Every - Chris Newton (comments tagged CSN)
 *
 * Notes:
 * 1. Using a very simple hardware setup consisting of
 *     a. IBT-2 motoir control board
 *          - requires only 2 PWM pins
 *          - enable always high or would require 3rd pin
 *     b. rudder sesnor
 *     c. No current/temp/voltage sensing implemented but 25amp breaker on motor line
 *     d. no rudder stop switches but implemented in software
 * 2. position(value)
 *    - this method controls the rudder position via 'value'
 *    - value dictates direction of rudder movement as well as its speed
 *    - the speed aspect is a bit unclear but I implemented acceleration that I think honors
 *      the original intent.
 * 3. it would be nice to have absolute movement commands, as in:
 *    - RUDDER CENTER, RUDDER_FAR_STBD, RUDDER_FAR_PORT
 *    - this would be easier to position the rudder when in manual mode.
 */

In terms of what is different between the Nano and Nano Every:

1. Clocks and interrupts
2. watchdog timer implementation
3. ADC implementation
4. pin locations for PWM
5. Might be easier to say, most everything!

I have tested this setup at the dock and it seems to work. The rudder moves in manual mode and the autopilot seems to work when egaged as well. Will let you know based on actual trials.

[motorNanoEvery20240327.zip](https://github.com/pypilot/pypilot/files/14779220/motorNanoEvery20240327.zip)

Regards,
Chris

circus37, 2024-07-19:
Hello i have the same problem but i can t upload nothing to the nano every ide arduino say error compilation permisssion denied  i see my card it on the rigth port my ide was 1.8.19 i whant to test your code thanks

DarfieldChris, 2024-07-19:
The code is in the ZIP attached to my original  comment.
Let me know if you cannot access the code.

circus37, 2024-07-19:
Hello really thanks i upload your code now i need to test

circus37, 2024-09-26:
Hi Chris i test pypilot whith the every but i have no communication on tx
rx pin do you test it or Just whith usb and another question the ibt2 is
wiring on d5 d6 like regular nano ? Thanks

circus37, 2024-09-27:
Hello i test your code but i have always port_pin_fault i dont know how to close this problem i think it bad pullup resitor

circus37, 2024-09-27:
Sorry 😅 i take my problem i soldering ruder sens wire on a2 not a0 haha

DarfieldChris, 2024-09-27:
Hi,

I used pins 5 and 6 for the IBT2. (the enable ws wired high permenantly,
which may be worth revisiting.).

Communication has always been via USB cable, I have not wired the tx and rx
lines directly.

I can also add that I have now put a few hundred miles on the autopilot and
it works!

Chris

DarfieldChris, 2024-09-27:
> Hi Chris i test pypilot whith the every but i have no communication on tx rx pin do you test it or Just whith usb and another question the ibt2 is wiring on d5 d6 like regular nano ? Thanks Le ven. 19 juil. 2024, 14:56, chevalier ronan ***@***.***> a écrit :
> […](#)

Hi,

Tried to email a response but it did not seem to work.

I used pins 5&6 for the IBT2. The enable pin was wired high, which might be worth switching high and lo as well.

I use a USB cable for communication, and have not tried the rx,tx pins directly.

Also, an update. I have used the autopilot for 200+ miles now and it works!

Chris

DarfieldChris, 2024-09-27:
> Sorry 😅 i take my problem i soldering ruder sens wire on a2 not a0 haha

Testing is always the hard part!

circus37, 2024-09-27:
Hello yes good New  now install in the boat good job thanks