pypilot/pypilot #226: Control buttons not responding on mobile / course can't be changed

matztam, 2024-04-06:
When ap is enabled the control buttons (-1 -1, +1, +10) do not respond to user interaction on mobile devices. 

How to reproduce:
1. Open pypilot webinterface in chrome
2. Enable Developer Tools (F12)
3. Toggle device toolbar (Ctrl+Shft+M)
4. Enable AP
5. Try to change the commanded course

Version:
Latest master ([c2b7797](https://github.com/pypilot/pypilot/commit/c2b7797c0dd5c9d0eafdfbe778360b0005ba3568))

seandepagnier, 2024-04-07:
It works using my latest... tinypilot image which should be in sync with master.     I tried on both a mobile device and in the debugger according to your instructions.

matztam, 2024-04-08:
Where can I get the tinypilot image?

seandepagnier, 2024-04-08:
Could you test it on a few different systems?   I want to reproduce the problem you have but cannot. 

This behavior did exist a few revisions back in git, but I since fixed it, so I am wondering if that is the reason.

matztam, 2024-04-08:
I tried on a different system today but was unable to reproduce it. 

The system on which the error occurred is on my boat. Unfortunately, I live some distance away. When I am back there (maybe next weekend) I can carry out further analysis.

seandepagnier, 2024-04-09:
there was a bug regarding this recently fixed.   Is it possible on that
system the latest code was not both pulled and installed??

matztam, 2024-04-09:
The system I tested on yesterday was definitely up to date. I was unable to reproduce the error there.
I thought the system on my boat was also on the latest version. I'll have to check again when I get back to the boat. Maybe the last update didn't go through?

seandepagnier, 2024-04-09:
yeah check it and confirm if possible,  that is the most likely explanation
but it would be good to know for sure.

matztam, 2024-04-15:
I have made sure that the latest version is installed. The problem persists. My pypilot.tcz is attached.
[pypilot.tcz.zip](https://github.com/pypilot/pypilot/files/14976129/pypilot.tcz.zip)

matztam, 2024-04-15:
After clearing the browser cache it works 🤦
Sorry for the trouble!

seandepagnier, 2024-04-15:
ok so it was caching the old version.  Makes sense, and thanks for
reporting this so the solution is verified.