pypilot/pypilot #229: Reducing accelerometer noise

mgrouch, 2024-04-20:
A lot of accelerometer noise comes from sounds. 
Is it possible to use some kind of fusion algorithm of microphone and accelerometer data to reduce the noise in accelerometer?
Or are there specific mounting instructions for accelerometer (sound proofing, padding on a mount) to reduce high frequency noises?

thanks

seandepagnier, 2024-04-22:
Do you mean audio from vibrations or electrical noise?

Regardless, the frequency should be so high to be irrelevant for the most part for inertial systems.

mgrouch, 2024-04-22:
> Do you mean audio from vibrations or electrical noise?
> 
> Regardless, the frequency should be so high to be irrelevant for the most part for inertial systems.

Mechanical vibrations from sound. They are high frequency. But they are not Gaussian white noise.