pypilot/pypilot #239: Arduino not detecting on pypilot

SoorajR-ai, 2024-08-26:
I have uploaded the motor code to arduino nano and connected to raspberry pi. when i open control in pypilot, its showing none.

seandepagnier, 2024-08-26:
try testing it by closing all pypilot processes and services.   Then run pypilot_servo on a console.   This is the simplest script to talk to the motor controller.

Ensure you have the serial device, probably /dev/ttyUSB0, and ensure pypilot probes it.  By default it should, but you might have to add it to ~/.pypilot/serial_devices if you are using openplotter.    Ensure the baud rate is possible.   Post the log output, or the console outputs.

A lot of other people have managed to do this, but it takes one small issue to make it not work.    Also consider trying the tinypilot image on a pi zero because it is a known working state.    Finally, consider connecting the uart directly to gpio on the pi without using usb (eventually)   My motor controllers do this with optical isolation.

SoorajR-ai, 2024-08-27:
I'm a dummy just trying to learn what this do. I only followed the tutorial to reach here. I dont know much more about these things.

SoorajR-ai, 2024-08-27:
ok, i did try to run pypilot_servo and this shows:

`pi@openplotter:/usr/local/bin $ ./pypilot_servo
pypilot Servo
world magnetic model not available
nmea process 6587
nmea server on port 20220: bind failed.
gps process 6590
loading servo calibration /home/pi/.pypilot/servocalibration
WARNING: using default servo calibration!!
pypilotServer process 6592
load file /home/pi/.pypilot/pypilot.conf
store_file /home/pi/.pypilot/pypilot.conf.bak 959.694 False
pypilot_server: bind failed; already running a server?
gps filter process 6591
signalk zeroconf service add openplotter._http._tcp.local. _http._tcp.local.
signalk zeroconf service add openplotter._http._tcp.local. _http._tcp.local.
signalk zeroconf service add openplotter._http._tcp.local. _http._tcp.local.
nmea server on port 20220: bind failed.
nmea server on port 20220: bind failed.
gpsd connected
GPSD devices []
nmea server on port 20220: bind failed.
pypilot_server: bind failed; already running a server?
nmea server on port 20220: bind failed.
nmea server on port 20220: bind failed.
servo probe ('/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0', 38400) 964.687765579
failed to open servo on: ('/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0', 38400) [Errno 16] could not open port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: [Errno 16] Device or resource busy: '/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0'
nmea server on port 20220: bind failed.
signalk process 6589
pypilot_server: bind failed; already running a server?
signalkread token eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJkZXZpY2UiOiJQWVBJTE9ULTg3ZWUtNGE3Mi04MGNhLWVkODNkYTRkNTkwYiJ9.m0I0Gxnabp98dufwa6N8nbXlPD6h42JFP2fPDVlB_yE
signalk server found 10.42.0.1:3000
nmea server on port 20220: bind failed.
signalk found ws://10.42.0.1:3000/signalk/v1/stream?subscribe=none
nmea server on port 20220: bind failed.
nmea probe ('/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0', 38400)
failed to open ('/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0', 38400) for nmea data [Errno 16] could not open port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: [Errno 16] Device or resource busy: '/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0'
nmea probe ('/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0', 4800)
failed to open ('/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0', 4800) for nmea data [Errno 16] could not open port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: [Errno 16] Device or resource busy: '/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0'
signalk connected to ws://10.42.0.1:3000/signalk/v1/stream?subscribe=none
nmea server on port 20220: bind failed.
pypilot_server: bind failed; already running a server?
nmea server on port 20220: bind failed.
servo probe ('/dev/ttyOP_gps0', 38400) 969.825171683
failed to open servo on: ('/dev/ttyOP_gps0', 38400) [Errno 2] could not open port /dev/ttyOP_gps0: [Errno 2] No such file or directory: '/dev/ttyOP_gps0'
nmea server on port 20220: bind failed.
nmea server on port 20220: bind failed.
pypilot_server: bind failed; already running a server?
nmea server on port 20220: bind failed.
nmea server on port 20220: bind failed.
nmea server on port 20220: bind failed.
pypilot_server: bind failed; already running a server?
nmea server on port 20220: bind failed.
nmea probe ('/dev/ttyOP_gps0', 38400)
failed to open ('/dev/ttyOP_gps0', 38400) for nmea data [Errno 2] could not open port /dev/ttyOP_gps0: [Errno 2] No such file or directory: '/dev/ttyOP_gps0'
nmea probe ('/dev/ttyOP_gps0', 4800)
failed to open ('/dev/ttyOP_gps0', 4800) for nmea data [Errno 2] could not open port /dev/ttyOP_gps0: [Errno 2] No such file or directory: '/dev/ttyOP_gps0'
nmea server on port 20220: bind failed.
nmea server on port 20220: bind failed.
pypilot_server: bind failed; already running a server?
nmea server on port 20220: bind failed.
^CProcess Process-6:
Process Process-3:
Traceback (most recent call last):
  File "/usr/local/bin/./pypilot_servo", line 33, in <module>
Process ZeroConfProcess-2:
    sys.exit(load_entry_point('pypilot==0.56', 'console_scripts', 'pypilot_servo')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pypilot/servo.py", line 824, in main
    time.sleep(dt)
KeyboardInterrupt
`

SoorajR-ai, 2024-08-27:
[image]