pypilot/pypilot #243: Use of BNO085 AR VR IMU

Fgonzalez22, 2024-10-01:
Hi, im not satisied with the response of my MPU9255. Does pypilot work with the BNO085 AR VR IMU?

https://es.aliexpress.com/item/1005007659429949.html?spm=a2g0o.cart.0.0.69bc7a9d9GDRHN&mp=1&gatewayAdapt=glo2esp

Thanks!

seandepagnier, 2024-10-01:
No it supports bno055, see:
https://github.com/seandepagnier/RTIMULib2

    Many MPU9255 are defective.   Did you try several?    The icm20948 is just as good but without many  defective ones.

What is wrong with the response?

Fgonzalez22, 2024-10-01:
Thanks Sean. I try another mpu same model. Will check if it works better. Calibration doesn't fit completely. Thought it might not be necessary for this model but I have up to 20degrees deviation while in navigating.