pypilot/pypilot #246: Possibly better Kalman filter than the one implemented by RTImuLib2

mgrouch, 2025-02-11:
RTImuLib2 Kalman filter model doesn't include estimation of the bias of gyro into its model.
It relies purely on calibration.

Kalman QMEKF filter which is published more recently might give better results as the bias is a part of the model.

See: Based on: https://github.com/thomaspasser/q-mekf

And version adapted for esp32

https://github.com/bareboat-necessities/bbn-wave-period-esp32/blob/main/bbn_wave_freq_m5atomS3/KalmanQMEKF.h

Thanks

seandepagnier, 2025-03-28:
It depends heavily on the inertial sensors and their intended use.

Yes it is great and interesting the estimate bias, but on a boat with the recommended sensors it is not really an issue because:
1)  the bias is compensated over time by lowpass filter and this works much better than aircraft or drones because the boat is never spinning continuously on any axis
2)  the sensors are actually pretty good and the bias does not drift that fast
3)  all of the logic is based on relative corrections, so absolute error in yaw gyro bias only leads to slight course error which the user will either compensate manually, or it gets handled by the outer control loop in nav mode.

But yes, sure it would be great to support additional filters.   Keeping the simpler known working solution is essential for reliability and understanding but an easy way to swap the filter on the fly would be ideal to compare performance.

mgrouch, 2025-03-29:
It grew on me the realization that there can be better Kalman model for marine purposes. 
A single Kalman model instead of chained Kalman filters. A Kalman model which fuses IMU, GPS, wind and speed through the water. Might be even temperature dependent biases model. In marine applications also it should take advantage on zero average vertical displacement. Compass headings should have Centripetal acceleration correction. 
Biases should be part of the model where possible. Wind calculated by the model should be corrected for mast height, movements, heel. 

Also support for 3 IMUs with detection of a fault on one is something I think would be very useful. 
And increased accuracy. 

Thanks

rgleason, 2025-03-29:
I really like these two:

1. Wind calculated by the model should be corrected for mast height, movements, heel.
2. Kalman model which fuses IMU, GPS, wind and speed through the water. (such that DR is continuously calculated even if GPS drops out, such that DR position can be reported to other plugins/modules)

seandepagnier, 2025-03-29:
There are corrections for mast height for wind.

Centripetal acceleration calculation is really difficult in waves.   This would work in higher speed boats in flat waters, but typical sail boats in waves it will be below the noise of the sensor.   A lot of these techniques are used in aircraft at much higher speeds without wave dynamics.   The water speed sensor on boats that would be used for this is not very accurate usually and at low speeds I dont really see this method adding much value, but interesting to explore.

I am not against these ideas, I would just always keep the basic model in place which uses the least amount of computation.

As for triple redundant, yes they can be mounted so that non axes are parallel.   This also allows automatic calibration of the sensor over different parameters on the fly and reduced noise as well as the fault detection.   It is my intention to support multiple pypilots with standby units ready to take over automatically with their own battery pack so they can continue steering for a few hours without the main power working.