pypilot/pypilot #248: NMEA parse failed when some fields are missing

rhbvkleef, 2025-02-18:
I recently installed a Pypilot-based autopilot on our boat, and until we hard-wire the thing in, we are using NMEA0183-over-TCP. While testing it with the simulation mode of our Lowrance HDS-5, I noticed that things weren't working, and after looking in the log file, it seems to be because of failing parsing of the NMEA input.

Specifically, the DPT parsing failed when it contains no data. I'd want it to fail a bit more gracefully, but I guess it's fine.

The VWH fails to parse when the water speed fields are missing. I guess the compass heading is still useful, so I would think that it should be able to decode the compass heading, and not pull out the speed?

Now the most problematic: the GPRMC sentence fails to decode probably due to the "UTC of position fix" field missing. I would suggest to use the current time in lieu of this field containing any data.

Here's a couple of lines from the logfile. These lines keep repeating:

```
2021-11-21_12:25:00.56247 exception parsing dpt could not convert string to float: '' $SDDPT,,,*7B
2021-11-21_12:25:00.56260 exception parsing vhw could not convert string to float: '' $SDVHW,273.1,T,278.7,M,,N,,K*4F
2021-11-21_12:25:00.56277 nmea failed to parse gps $GPRMC,,A,2545.1062,N,08007.5374,W,20.0,273.2,01014294967283,5.5,W,S*3C could not convert string to float: ''
```

Here's a truncated NMEA-0183 input:

```nmea
$GPRMB,A,0.000,L,SIMWPT,SIMWPT,2545.1464,N,08008.3372,W,0.82,273.2,20.0,V,S*63
$SDHDG,278.7,,,5.5,W*2D
$GPXTE,A,A,0.000,L,N,S*21
$SDDBT,,,,,,*45
$SDDPT,,,*7B
$SDMTW,,C*1A
$SDVLW,,N,,N,$SDVLW,,N,,N,5363.4,N,$SDVLW,,N,,N,$SDVLW,,N,,N,5363.4,N,2799.8,N*55
$SDVHW,273.1,T,278.7,M,,N,,K*4F
$WIMWD,,T,,M,,N,,M*5A
$WIMWV,,R,,N,V*34
$WIMWV,,T,,N,V*32
$IIXDR,C,C,,C,AIRTEMP,A,,D,HEEL,A,,D,TRIM,P,,B,BARO,A,A,,D,RUDDER*05
$GPGGA,,2545.1006,N,08007.4264,W,1,00,,,M,,M,,*44
$GPGLL,2545.1006,N,08007.4264,W,,A,S*4D
$SDHDG,278.7,,,5.5,W*2D
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$SDHDG,278.7,,,5.5,W*2D
$GPRMC,,A,2545.1006,N,08007.4264,W,20.0,273.2,01014294967283,5.5,W,S*3F
$SDHDG,278.7,,,5.5,W*2D
$SDHDG,278.7,,,5.5,W*2D
$GPVTG,273.2,T,278.7,M,20.0,N,37.0,K,S*39
$SDHDG,278.7,,,5.5,W*2D
$SDHDG,278.7,,,5.5,W*2D
$GPZDA,,01,01,-13,-02,00*48
$GPAAM,V,V,0.00,N,SIMWPT*22
$SDHDG,278.7,,,5.5,W*2D
$GPAPB,V,V,0.000,L,N,V,V,273.2,T,SIMWPT,273.2,T,273.2,T,S*6D
$GPBOD,273.2,T,278.7,M,SIMWPT,SIMWPT*49
$SDHDG,278.7,,,5.5,W*2D
$GPBWC,,2545.1464,N,08008.3372,W,273.2,T,278.7,M,0.82,N,SIMWPT,S*5A
$SDHDG,278.7,,,5.5,W*2D
$GPBWR,,2545.1464,N,08008.3372,W,273.2,T,278.7,M,0.82,N,SIMWPT,S*4B
$GPRMB,A,0.000,L,SIMWPT,SIMWPT,2545.1464,N,08008.3372,W,0.82,273.2,20.0,V,S*63
$SDHDG,278.7,,,5.5,W*2D
$GPXTE,A,A,0.000,L,N,S*21
$SDDBT,,,,,,*45
$SDDPT,,,*7B
$SDMTW,,C*1A
$SDVLW,,N,,N,$SDVLW,,N,,N,5363.5,N,$SDVLW,,N,,N,$SDVLW,,N,,N,5363.5,N,2799.8,N*55
$SDVHW,273.1,T,278.7,M,,N,,K*4F
$WIMWD,,T,,M,,N,,M*5A
$WIMWV,,R,,N,V*34
$WIMWV,,T,,N,V*32
$IIXDR,C,C,,C,AIRTEMP,A,,D,HEEL,A,,D,TRIM,P,,B,BARO,A,A,,D,RUDDER*05
$GPGGA,,2545.1010,N,08007.4325,W,1,00,,,M,,M,,*47
$GPGLL,2545.1010,N,08007.4325,W,,A,S*4E
$SDHDG,278.7,,,5.5,W*2D
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$SDHDG,278.7,,,5.5,W*2D
$GPRMC,,A,2545.1010,N,08007.4325,W,20.0,273.2,01014294967283,5.5,W,S*3C
```

seandepagnier, 2025-02-22:
Which version of pypilot are you running?   It might be several years out of date?

Anyway I do not get the errors except I do for the VHW sentence which makes sense because your sentence does not have water speed in it.   pypilot expects water speed in this sentence which may be a mistake as in your case the sentence is used for heading.   pypilot is not going to use an external heading because it relies on its own IMU to determine heading.

rhbvkleef, 2025-04-05:
That might explain it, yes. Euhh I'm indeed thinking my PyPilot install hasn't been updated for a while.

closed 2025-04-05 by rhbvkleef