pypilot/pypilot #249: wind height compensation

mcmonagle101, 2025-03-30:
Where is an issue in sensor.py class BaseWind with the height compensation.  Normally I would only report after a real test but that will be in a couple months.  Feel pretty confident that what follows is pretty close to being correct.  There are several ways to fix this but need to use the same units.   I'm assuming speed is nautical miles and angles are degrees.
Also need to document what units are used for compensation_height.
I'm going to assume meters.

Here is the part of the code in question:            speed = data['speed']
            direction = data['direction']
            dx = speed*math.sin(math.radians(direction))
            dy = speed*math.cos(math.radians(direction))

above is correct  and dx and dy are nautical miles per hour
The math radians in the next line bothered me but realize it really converting rollrate and pitchrate  to radians from degrees.
            m = math.radians(self.compensation_height.value)
As far as I can tell people are putting in height in meters. That is OK but the radius in the next calculation needs to be in nautical miles.  Also it may seem strange but rollrate and pitchrate need to be converted to radians/hour.
            dx -= m * self.boatimu.SensorValues['rollrate'].value # is this positive or negative?!?
            dy -= m * self.boatimu.SensorValues['pitchrate'].value
The rest is fine.
            data['speed'] = math.hypot(dx, dy)
            data['direction'] = math.degrees(math.atan2(dx, dy))
........................................
To fix in a more mathematically obvious form I'd in "m" calculation convert to nautical miles  and in the second dx ,dy  do the radian and hour conversion.

It's more efficient to to do it all in the m calculation.
Conversion constants to 3 significant figures
a= degree to radian =  .0174
b= rad/sec to rad/hour =3600 
d= meters  to nautical miles = 0.00054

if in meters:
m=  self.compensation_height.value * a * b * d
or
m=  self.compensation_height.value * 0.0338

To compensate with the code as is:
Set  height in meters * (b*d | 1.94)
Just multiple height in meters by 1.94

Hope that make sense. If not let me know.

rgleason, 2025-04-01:
Is this compensation for wind height, due to the friction of the earth/waves etc and the fact that wind speeds are higher further from sealevel?   If wind speed is adjusted for height, what is the height that the new wind speed is calculated for?

Is there also calculation for the affect of roll and pitch on wind speed in pypilot?

rgleason, 2025-04-02:
I asked if  "Is there also calculation for the affect of roll and pitch on wind speed in pypilot?"

Sean while I have your attention would you please do me a favor and close the 6 (of 7) issues that I noted to be closed in weather-routing?  https://github.com/seandepagnier/weather_routing_pi/issues

We are trying to triage the many duplicate and old Issues and have consolidated these into https://github.com/rgleason/celestial_navigation_pi/issues   and Sebastien and I are busy whittling them down.

mcmonagle101, 2025-04-02:
I keep wondering if under compensating is better also. That would better reflect what the sails have to deal with but full compensation should keep the autopilot happier.  At some point with light wind and large waves it's challenging to sail no matter what you do.

Did some testing at home playing a nmea file in opencpn and verified that compensation is being done.
Of course have no idea if it's correct.
It will be middle of May before I can test for real.

If one just drifts with waves and reasonably steady wind should be obvious if  you watch heading  in wind mode.
Have you heard any reports from people trying height compensation?

I think the sign is correct, but if not correct could make the height negative, but it may be changing entering it.

For this calculation the dx and dy  are with respect to the wind sensor. 
As to heel adjustment:
Is there some benefit to have a better true wind calculation for other calculations?
I don't think it matters upwind  for steering.  
Maybe calculate how much error for some directions and likely  large heel angles  that one would really sail with.
Max speed error would be on beam.
Max direction error I think would be  at 45 and 135. Less certain on that.