pypilot/pypilot #250: [FR] Use satellite compass instead of IMU as source of heading

mgrouch, 2025-04-14:
They are getting cheaper and not subject to magnetic interference 
 
Example:

https://holybro.com/products/h-rtk-unicore-um982

UM982. Can work without RTK just by using two antennas. Produces regular NMEA 0183 stream with some added sentences for Heading, best position estimate and orientation. With high rate. 

Thanks

mgrouch, 2025-04-15:
Looks like it’s easily configurable via NMEA 

https://github.com/orgs/bareboat-necessities/discussions/514

Error 0.4 m for position with dual antenna mode without even using RTK. 

Error for yaw (heading) is 0.1 degrees with 1m distance between antennas

piranha771, 2025-08-22:
Did you get this working with pypilot? Which NMEA sentences are required?. I just try to do by shoving everything into signalK with acting as a "magnetic compass" but no luck so far getting it recognized from pypilot.