pypilot/pypilot #255: Where can I get a motor controller in the US?

Rudder2, 2025-05-26:
Where can i get a replacement motor controller in the US? PyPilot.org store is 404. Need a replacement motor controller in 2 weeks. 

I should of replaced that diode that protection the motor controller the first time...Doh!

leoniedu, 2025-05-26:
Perhaps here ?
https://www.navitop.fr

Rudder2, 2025-05-27:
@leoniedu That's not a US location, thank you, could order it from OpenMarine if I want to order from outside US. Also, they aren't the same as the one Sean makes. The only part that matches up is the data cable and the drive connection and the clutch. 

I have every option that Sean offered and the plugs he used for all the options. I wish I could buy it from Sean because I would rather support the creator but his ordering site is down.

rgleason, 2025-05-29:
I think I got his last one.