pypilot/pypilot #257: Compass software issue

Fgonzalez22, 2025-07-07:
Hi, I have a problem with the calibration of the compass. 
As shown in the picture the window does not show any data when turning the Imu. 
Accelerometer cal and boat leveling when’s ok. 
Imu is a icm-20948 from adafruit
Software is up to date. All openplotter and pypilot updates v4.0.11
Raspi 5

I have a second raspi with the same installation and the sensor and calibration works well there. 

I have uninstalled and reinstalled pypilot in case there was something corrupt but the problem persists. 

Any suggestions would be useful. I’m really stuck

Thanks!

[image]

Fgonzalez22, 2025-07-12:
Update, I got this lines for pypilot_boatimu that show problems

i@openplotter:~ $ pypilot_boatimu
pypilotServer process 7008
imu process 7006
made imu process realtime
Using settings file RTIMULib.ini
Settings file RTIMULib.ini loaded
load file /home/pi/.pypilot/pypilot.conf
store_file /home/pi/.pypilot/pypilot.conf.bak 1589.554 False
pypilot_server: bind failed; already running a server?
Detected ICM20948 at standard address
Using fusion algorithm Kalman STATE4
IMU Name: ICM-20948
min/max compass calibration not in use
Using ellipsoid compass calibration
Using accel calibration
AK09916 failed to set rate!!!!!
ICM-20948 init complete
ICM20948 bad data 3 7
failed to read IMU! 1591.142439015
Detected ICM20948 at standard address
IMU Name: ICM-20948
min/max compass calibration not in use
Using ellipsoid compass calibration
Using accel calibration
ICM-20948 init complete
imu process failed to keep time 0.4421035170000778 1591.142437735 1591.58443364 1591.584484357 1591.584541122
ICM20948 bad data 0 9
failed to read IMU! 1591.584575867
Detected ICM20948 at standard address
IMU Name: ICM-20948
min/max compass calibration not in use
Using ellipsoid compass calibration
Using accel calibration
ICM-20948 init complete
pypilot_server: bind failed; already running a server?
imu process failed to keep time 0.42834243900006186 1592.184694249 1592.612987741 1592.61300716 1592.613036521
failed to read IMU! 1592.71308911141.1455303474702 heading 17.771965150803133 
Detected ICM20948 at standard address
IMU Name: ICM-20948
min/max compass calibration not in use
Using ellipsoid compass calibration
Using accel calibration
ICM-20948 init complete
imu process failed to keep time 0.4329556379998394 1593.213233763 1593.646157615 1593.646174009 1593.646189197
ICM20948 bad data 3 10
failed to read IMU! 1593.646227029
calibration loaded, starting 7004
Detected ICM20948 at standard address
IMU Name: ICM-20948
min/max compass calibration not in use
Using ellipsoid compass calibration
Using accel calibration
ICM-20948 init complete
imu process failed to keep time 0.42954251400010435 1594.246371449 1594.675883015 1594.67589796 1594.675913815
ICM20948 bad data 10 11
failed to read IMU! 1594.675956964
Detected ICM20948 at standard address
IMU Name: ICM-20948
min/max compass calibration not in use
Using ellipsoid compass calibration
Using accel calibration
pypilot_server: bind failed; already running a server?
ICM-20948 init complete
imu process failed to keep time 0.428924256000073 1595.276058291 1595.704948672 1595.704965822 1595.704982398
IMU all sensor axes verifiedoll 141.11214279251013 heading 12.79100615445714 
pypilot_server: bind failed; already running a server?ing 13.034049661071458  
pypilot_server: bind failed; already running a server?ding 13.114005960719437 
^CProcess Process-3:92491 roll 140.48923812781774 heading 13.167971962777802  
Traceback (most recent call last):
  File "/usr/local/bin/pypilot_boatimu", line 33, in <module>
    sys.exit(load_entry_point('pypilot==0.56', 'console_scripts', 'pypilot_boatimu')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pypilot/boatimu.py", line 613, in main
Process Process-1:
Process Process-2:
    time.sleep(dt)

pwagsc, 2025-08-05:
is the failing pi a 5 and the other pi a 4?  There were some issues with the Adafruit i2c driver on 5.  Make sure you have it updated.  Make sure it says it inited properly in the summary at boot if using openplotter.

luckybeanz, 2026-11-01:
This should not be closed.   I have been trying for months to get pypilot working on a pi5 as the documentation sates it works.   It does not. 

failed to read IMU!
ICM20948 bad data 0 10
Detected ICM20948 at standard address
IMU Name: ICM-20948
ICM-20948 init complete
imu process failed to keep time