pypilot/pypilot #258: Averaging angles looks wrong

mgrouch, 2025-07-07:
I looked at heading_filter function as example. 
It applies low pass on angle value. Angle value is a circular parameter. I think it’s invalid way to average angles. 

Correct way of averaging angles is low pass on sin and cos, the normalization and reverting back to angle with atan2. 

It seems everywhere in the code averaging of angles is not right. 

Thanks

mgrouch, 2025-07-07:
- **Example:**  
     - `a = -170°`, `b = 190°`, `lp = 0.5`.  
     - The function sees `a - b = -170 - 190 = -360` (not `> 180`), so **no adjustment is made**.  
     - The average is `(-170 + 190)/2 = 10°`.  
   - **Expected behavior:**  
     - `-170°` is equivalent to `190°`, and `190°` is equivalent to `-170°`.  
     - The true average should be `10°` (correct here), but if `a = -170°` and `b = 170°`:  
       - The function computes `(-170 + 170)/2 = 0°`, but the **correct circular average is `180°`** (since `-170°` and `170°` are both near the `180°` line).

mgrouch, 2025-07-07:
| **Case**                | **Input Angles (`a`, `b`)** | **Original Result** | **Correct Short-Path Result** | **Error** | **Why Original Fails** |
|-------------------------|----------------------------|---------------------|-------------------------------|-----------|------------------------|
| **Near-Opposite (Long Path Bias)** | `10°`, `170°`  | `154°`              | **262.6°**                    | **+108.6°** | Takes long-path average instead of short-path. |
| **Wrap-Around (359°→1°)** | `359°`, `1°`             | `19.8°`             | **0.2°**                      | **-19.6°** | Misadjusts `359°` to `-1°`, skewing average. |
| **Unnormalized (Negative Angle)** | `-10°`, `190°` | `172°`              | **184°**                      | **+12°** | Fails to normalize `-10°` to `350°`. |
| **Floating-Precision Edge** | `0°`, `179.999°` | `162°`           | **270°** (approx.)            | **+108°** | Detects as `<180°`, picks long path. |
| **Large Gap (No Wrap)** | `45°`, `315°`            | `288°`              | **0°**                        | **-72°** | Incorrectly treats as non-wrapping case. |

seandepagnier, 2025-07-08:
Is this the heading_filter in boatimu?

If so the only real bug is the testing "if not a..."  since passing 0 will cause the filter to not work exactly right but it is a minor bug.   I believe those lines can be removed.

mgrouch, 2025-07-08:
Something like this would work

```python

import math

class AngleEstimate:
    def __init__(self, angle=0.0, magnitude=1e-12, variance=(math.pi**2)/4.0, consistency=0.0):
        self.angle = angle              # Filtered angle in degrees
        self.magnitude = magnitude      # Confidence (0–1)
        self.variance = variance        # Estimated circular variance (rad²)
        self.consistency = consistency  # Dot product similarity [0–1]

class AngleAverager:
    def __init__(self, alpha_init=0.004):
        """
        Create an AngleAverager.
        Use alpha >= 0 for fixed smoothing,
        Use alpha < 0 for adaptive smoothing based on variance.
        """
        self.alpha = alpha_init
        self.initialized = False
        self.angle_prev = 0.0
        self.variance_prev = (math.pi ** 2) / 4.0

    def reset(self, angle_deg):
        """Explicitly set initial angle and reset variance."""
        self.angle_prev = angle_deg
        self.variance_prev = (math.pi ** 2) / 4.0
        self.initialized = True

    def average360(self, new_angle_deg):
        """Filter angle in [0, 360) degrees (e.g., compass heading)."""
        if not self.initialized:
            self.reset(new_angle_deg)

        cx, cy = self._polar_deg(self.angle_prev)
        nx, ny = self._polar_deg(new_angle_deg)

        new_mag = self._magnitude(nx, ny)
        new_var = self._estimate_variance(new_mag)
        eff_alpha = self.alpha if self.alpha >= 0 else self._adaptive_alpha(self.variance_prev, new_var)

        fx = (1.0 - eff_alpha) * cx + eff_alpha * nx
        fy = (1.0 - eff_alpha) * cy + eff_alpha * ny

        filtered_angle = math.degrees(math.atan2(fy, fx)) % 360.0
        mag = self._magnitude(fx, fy)
        var = self._estimate_variance(mag)
        cons = self._consistency(cx, cy, nx, ny)

        self.angle_prev = filtered_angle
        self.variance_prev = var

        return AngleEstimate(filtered_angle, mag, var, cons)

    def average180(self, new_angle_deg):
        """Filter angle in [0, 180) degrees (e.g., wave direction without sign)."""
        if not self.initialized:
            self.reset(new_angle_deg)

        cd = 2.0 * self.angle_prev
        nd = 2.0 * new_angle_deg

        cx, cy = self._polar_deg(cd)
        nx, ny = self._polar_deg(nd)

        new_mag = self._magnitude(nx, ny)
        new_var = self._estimate_variance(new_mag)
        eff_alpha = self.alpha if self.alpha >= 0 else self._adaptive_alpha(self.variance_prev, new_var)

        fx = (1.0 - eff_alpha) * cx + eff_alpha * nx
        fy = (1.0 - eff_alpha) * cy + eff_alpha * ny

        filtered_angle_doubled = math.degrees(math.atan2(fy, fx)) % 360.0
        filtered_angle = (0.5 * filtered_angle_doubled) % 180.0

        mag = self._magnitude(fx, fy)
        var = self._estimate_variance(mag) / 4.0  # Correct for doubling
        cons = self._consistency(cx, cy, nx, ny)

        self.angle_prev = filtered_angle
        self.variance_prev = var

        return AngleEstimate(filtered_angle, mag, var, cons)

    def get_angle(self):
        return self.angle_prev

    def get_variance(self):
        return self.variance_prev

    def is_initialized(self):
        return self.initialized

    @staticmethod
    def _polar_deg(angle_deg):
        rad = math.radians(angle_deg)
        return math.cos(rad), math.sin(rad)

    @staticmethod
    def _magnitude(x, y):
        return math.hypot(x, y)

    @staticmethod
    def _estimate_variance(mag):
        if mag <= 0.0:
            return math.pi ** 2
        if mag > 0.999:
            return 0.0
        return -2.0 * math.log(mag)

    @staticmethod
    def _consistency(x1, y1, x2, y2):
        dot = max(-1.0, min(1.0, x1 * x2 + y1 * y2))
        return 0.5 * (1.0 + dot)

    @staticmethod
    def _adaptive_alpha(var_current, var_new, eps=1e-6):
        a = var_new / (var_current + var_new + eps)
        return max(0.0, min(1.0, a))

````

mgrouch, 2025-07-08:
This line is problematic

```python
data['heading_lowpass'] = heading_filter(llp, data['heading'], self.SensorValues['heading_lowpass'].value)
````

due to heading_filter implemented incorrectly with low pass of circular angular values.

heading_lowpass value is read and propagated with incorrect values in many places of the code.

fixing heading_filter should be enough.

I haven't found any other places where average of angular is calculated but I can't guarantee that I didn't miss any

seandepagnier, 2025-07-08:
Your proposed solutions all use trig functions atan2 and are very complicated.

   I dont need to use that, yes I can average the values even if they are circular, it does work.    Did you try actually running the function?   I ran it makes the correct results for all the test cases you posted and not the values you claim it produces.

I agree it is broken for values of exactly zero but in practice this is a subtle issue.