pypilot/pypilot #259: SignalK: How to get recognized by pypilot as IMU?

piranha771, 2025-08-22:
I tried sending 5Hz `navigation.headingMagnetic`,` navigation.attitude`, `navigation.rateOfTurn` in every combination possible. It won't get recognized as an IMU sensor/input. 

But, if I for example send `navigation.courseOverGroundTrue`, `navigation.speedOverGround` and `navigation.position` pypilot is happily satisfied and detects it as a GNSS module immediately.

What I want todo: I have a working RTK moving base setup AKA dual GNSS with heading information. I want to put the heading info over as a IMU input over SignalK.

seandepagnier, 2025-08-24:
Yes, it essentially steers using the internal IMU and uses gps heading to update the internal heading to follow.

This might be the best starting point for using gps heading,  to simply run in gps mode, but it will still require compass calibration.

There is an experimental "gps pilot" in pypilot/pilots/gps.py  which really would have a lot more potential using gps heading if it were highly accurate with dual antennas and that is probably the right way to use this in the immediate future.

piranha771, 2025-09-22:
> So what I am afraid will happen, is people will wire an IMU into some device, send signalk over wifi to the autopilot and try to use this gyro and end up with terrible performance because they dont want to just stick a $10 IMU on the pilot itself. I am sorry if it seems annoying, but this is a real concern, but of course the source is open so I don't strictly prevent someone from doing it.

I totally understand this. And I see why my approach of using SignalK for this was wrong. I have now made a dirty hack: I ripped out most of IMU class. The IMU I have has it's own calibration system and outputs filtered data (witmotion HWT905). So I 
- take the True-Heading-GPS data from the F9P RTK modules via UBX protocol (they don't have NMEA $HDT sentences :( )
- convert true-heading to magnetic heading 
- replace the IMU MEMS-heading own heading
- feed this modified IMU data into BoatIMU
 
up on first sight this works very well. 

For the future I want to make a cleaner solution though. Maybe we can work something out to allow for HDT as an alternate way that does not use gpspilot (I understand that this works on getting direction from motion?)